MarlinFirmware/Marlin #11858: 1.1.x-bugfix AnyCubic Kossel Pulley delta configs endstops crash

F0x06, 2018-09-17:
### Description
During testing of the provided Anycubic Kossel Pulley config in `1.1.x-bugfix`, following the README, when i do `Set delta height` with Z-Probe V2 connected, my printer crashes in the XYZ MAX endstops, forcing me to shutdown the printer.

### Steps to Reproduce

1. Configure the firmware as described in README
2. Flash-it
3. Do "Set delta height with the Z-Probe version 2"

**Expected behavior:** [What you expect to happen]

Correct delta height around `333.50`

**Actual behavior:** [What actually happens]

Moves around 30mm down then goes backwards and crash on endstops

```
SENDING:M119

>>>> Tested all endstops, works correctly <<<<
Reporting endstop status
x_max: TRIGGERED
y_max: TRIGGERED
z_min: open
z_max: TRIGGERED

echo:enqueueing "G33 P1"
G33 Auto Calibrate
Checking... AC
.Height:320.00                                Offset:-16.80
echo:busy: processing
echo:busy: processing
echo:busy: processing
Calibration OK                                std dev:0.000
.Height:29.19                                Offset:-16.80 <<<< Wired height...

>>>> Crash to endstops happened here <<<<
```

#### Additional Information

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/2390437/Configs.zip)

I can help you to test these configs, also there is my hand-made working config that i'm currently using
[handmade_configs.zip](https://github.com/MarlinFirmware/Marlin/files/2390445/handmade_configs.zip)

Thank you in advance, F0x.

brandstaetter, 2018-09-17:
Try the bugfix-2.0.x version, I fixed a lot of issues there.

F0x06, 2018-09-18:
Hi @brandstaetter, same issue on bugfix-2.0.x.

1. Copied `./Marlin/src/config/examples/delta/Anycubic/Kossel` to `./Marlin`
2. Changed `ANYCUBIC_PROBE_VERSION` to `2`
3. Flashed firmware
4. Initialized EEPROM from LCD menu
5. Set Delta Height

brandstaetter, 2018-09-18:
Does homing work?
It kind of sounds like you inverted your stepper directions, has it worked with another firmware before? Can you try homing and then moving the Z direction?

F0x06, 2018-09-18:
Yes homing works. And yes worked on my own previous config (attached) I'll try to move Z from LCD menu

F0x06, 2018-09-18:
@brackendawson Connected with the back wire to the exterior side, like other endstops, these are working perfectly. Tell me when you can check on your printer :)

F0x06, 2018-09-19:
@brackendawson I successfully calibrated my priner, all is working properly now, great configs :) !

F0x06, 2018-09-19:
@brandstaetter Probably me, can i have a picture of your end-stops pins ? Maybe my probe has been wrongly assembled... In your build, the black wires are on which side ? Exterior (Close to the third unused pin) or Interior ? In my case Exterior like this

```
B: Black
R: Red
Z+ Z- Y+ Y- X+ X-
R   R   R   *   R   * 
B   B   B   *   B   *
*    *   *   *   *    *
```

brandstaetter, 2018-09-23:
I just looked in the build manual, it does say to connect it the same as the others?

b0ssman, 2018-09-23:
v1 probe is NO
v2 probe is NC
see https://github.com/emilio1625/Anycubic-Kossel-Marlin/blob/1.1.x/Marlin/example_configurations/delta/ANYCUBIC/Configuration.h

b0ssman, 2018-09-23:
also the v2 z offset is 15.9. thats from the official anycubic firmware.

F0x06, 2018-09-23:
@b0ssman On my printer with my V2 probe my offset is at -16.80, I have good results

brandstaetter, 2018-09-23:
With #11899 the risk to the probe should be resolved and it can be measured safely from the LCD UI

F0x06, 2018-09-23:
Nice I'll try it tonight :)

brandstaetter, 2018-09-23:
@F0x06 if you still have problems with the Z probe being connected wrong, just change the version to 1, that should help

F0x06, 2018-09-23:
@brandstaetter Thank you! I'll try it :)

F0x06, 2018-09-24:
@brandstaetter Probes are always miss configured in latest version, as said by @b0ssman
v1 probe is NO
v2 probe is NC

Or its me ?

```
// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#if ANYCUBIC_PROBE_VERSION == 1
  #define Z_MIN_PROBE_ENDSTOP_INVERTING false  // V1 Probe is NC
#elif ANYCUBIC_PROBE_VERSION == 2
  #define Z_MIN_PROBE_ENDSTOP_INVERTING true  // V2 Probe is NO
#endif
```
It sould be
```
// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#if ANYCUBIC_PROBE_VERSION == 1
  #define Z_MIN_PROBE_ENDSTOP_INVERTING true  // V1 Probe is NC
  #define Z_MIN_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.
#elif ANYCUBIC_PROBE_VERSION == 2
  #define Z_MIN_PROBE_ENDSTOP_INVERTING false  // V2 Probe is NO
  #define Z_MIN_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#endif
```

Thanks, otherwise it works really great, and the multi-probing is a lot faster now :)

brandstaetter, 2018-09-24:
@F0x06 please make the pull request yourself. I thought I have V2 Probes and it works for me as-is. I will recheck my wiring and also check with the other probes that I have. 

Plus, I finally need to build up the Linear Plus that I got to verify that setup too.

F0x06, 2018-09-24:
@brandstaetter OK, I'll check with the anycubic stock firmware before. if it's reversed too then its my probe, maybe wires are reverses by a manufacture error. I'll also ask on the FB group of others have the same issue.

brandstaetter, 2018-09-26:
@F0x06 I now built a Kossel Linear Plus I had sitting in a box for some weeks now - it seems to have a V1 probe, and the logic is really inverted to the V2 one...
Small fix is necessary, see #11929

F0x06, 2018-09-26:
@brandstaetter Nice!, I checked also on my side with Spike's Lee version, it's inverted. No need to make a pull request tonight so. Thank you for the fix. Feel free to close the issue when done :)

JelmerT, 2018-10-13:
Just bought a V2 probe, and it is definitely NC. 
So `Z_MIN_ENDSTOP_INVERTING` needs to be `false` for a V2 probe.
#11929 still has it reversed.

By the way, reversing wires on a simple switch doesn't make the switch behave differently. So it doesn't matter where the red and black wire go, as long as one is on the GND pin (middle row) and the other one is on the GPIO pin (top one).
I can make new PR if desired.

F0x06, 2018-10-13:
For V2 probe it should be
```
#define Z_MIN_PROBE_ENDSTOP_INVERTING false
 #define Z_MIN_ENDSTOP_INVERTING false
```

F0x06, 2018-10-13:
I'll make the PR :)

F0x06, 2018-10-14:
@thinkyhead @brandstaetter I guess the bug is still present on `1.1.x` and`on `1.1.x-bugfix` do you want me to make a PR also ? :)

brandstaetter, 2018-10-14:
is there a config for the kossel on 1.1.x?

JelmerT, 2018-10-14:
I ran into the bug when running `1.1.x` would be great if it got fixed there too

F0x06, 2018-10-23:
I'll do the PR for 1.1.x maybe tomorrow @JelmerT

JelmerT, 2018-10-24:
@F0x06 looks like #12101 already fixed this in the bugfix branch

F0x06, 2018-10-24:
@JelmerT you're right, so I can close this issue :)

closed 2018-10-24 by F0x06