MarlinFirmware/Marlin #11867: [2.0.x, Question] Is there any successful port to MBED?

dmachaty, 2018-09-18:
Hello.
Could you please point me to any build-passing Marlin 2.0.x fork, using mbed framework instead of arduino? I've tried to search, but didn't really find anything. I see this as a logical option when porting Marlin 2.0.x to a new CPU, but there's quite a lot of things to "work around" just to build with all libraries excluded. I would really appreciate some inspiration (just porting serial.print() took me an hour, and I'm pretty sure it's not gonna work as intended :-) ).

I'm trying to create a new board based on NXP MK64F chip, FRDM-K64F dev board. 

Current platformio.ini setup:

```
[env:frdm_k64f]
platform        = freescalekinetis
framework       = mbed
board           = frdm_k64f
build_flags     = ${common.build_flags} 
src_build_flags = --std=c++11
  -Wno-expansion-to-defined
lib_deps        = ${common.lib_deps}
lib_ignore      = 
  U8glib
  LiquidCrystal_I2C
  LiquidCrystal
  NewliquidCrystal
  LiquidTWI2
  Adafruit NeoPixel
  TMC2130Stepper
  Servo(STM32F1)
  TMC26XStepper
  U8glib-HAL
  TMC2208Stepper
  c1921b4
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_FRDM_K64F>
#monitor_speed = 250000 
```

Thank you for any advices.

thinkyhead, 2018-09-18:
I'm not aware of any. We are only using CMSIS at this time.
https://forums.mbed.com/t/cmsis-vs-mbed-os/1164

AnHardt, 2018-09-18:
Would be much easier if there would be a Arduino for mbed. But i'm not aware of one. Likely because everyone if feared from the overhead of the additional layer.

p3p, 2018-09-18:
I have had the LPC176x platform compile on top of mbed (during initial dev work long ago) but in the end I was using very little of the mbed features other than the build and I think I moved to CMSIS because the mbed framework can only build for the LPC1768 @ 96Mhz and not the LPC1769, (at least at the time not sure about now) 

There are a few issues to work around such as:
Slow IO, there are third part libraries that can improve that significantly, though for pulse generation you may need direct access.
Enums, Marlin is not fond of enums because a lot of constants need to be used by the preprocessor, mbed uses enums for pin identifiers which will need indirection,

dmachaty, 2018-09-19:
@thinkyhead - ok, thanks

@AnHardt - well, in a way, mbed looks pretty similar to arduino.. When I started peeking into the marlin/mbed thing, I obtained mainly undeclared function errors, lot of which were solved just by redefining function call. Which doesn't of course mean it would work well after compilation.. Whole another thing is all the "arduino-specific" stuff in Marlin, like specifically putting const strings into progmem (gcc usually puts all consts into flash, so pstr() and f() macros can be directly defined out).

@p3p - oh, ok, this seems like reasonable arguments. If the speed is an issue, I cannot object.. I'm just a little worried about using directly CMSIS, because it means a lot of work just to replace the mbed features (serial.printf for start). I've done some bare metal code before only for smaller chips..

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-15 by p3p

DerAndere1, 2019-10-23:
In case someone is still interested: The Arduino software framework was ported to run on mbed recently: http://blog.janjongboom.com/2019/08/01/arduino-mbed.html . This could make it easier to run Marlin 2.0 on mbed systems.

AnHardt, 2019-10-23:
> The Arduino software framework was ported to run on mbed

The other way around. Arduino now is able to compile code using the mbed-framework. It's not the Arduino IDE compiled running the embed-framework (running on the ARM processor).

AnHardt, 2019-10-23:
Sorry reading more it seems to be about compiling Arduino libraries used in the embed framework/environment. Or both?

However. Telling PlatformIO to use the mbed-toolchain to compile Marlin with a HAL for mbed seems to be possible.

But is that a good idea? Likely this approach will add overhead from Arduino librayes to that from mbed, plus the needed glue from Marlin. For the AVRs we tried to do the opposite - bypassing the Arduino framework to get more speed (v.e. fastio, reading ADCs, timer programming, modified fatfs to use fastio, customized software SPI routines everywhere, ...)