MarlinFirmware/Marlin #11868: Inductive Bed Sensor M851 and M206

totalretribution, 2018-09-18:
### Description

I have recently fitted an Inductive Sensor to my ender 3 and I am very confused about M851 and M206.

When my bed homes my nozzle is 0.3mm from the bed, I thought that M851 was used to then correct for this offset however if I set M851Z-0.3 and re-home, the printer homes correctly but then moves the bed by what appears to be 0.3mm. It then sets the current co-ordinates to Z0.3.

This means that now the nozzle is 0.6mm from the bed and if I move the bed to Z0 the nozzle is actually 0.3mm from the bed.

I then tried setting M851 to Z0 and used M206 and this almost gives me what I want in that after homing it adjusts the co-ordinates to be Z0.3 however it also changes the software endstops to Z0.3 and so I now can not move the nozzle any closer than 0.3mm to the bed.

So now the only way I can get the printer to work correctly is to physically adjust the inductive sensor so that it triggers when the nozzle actually touches the bed which is not desired especially since its very difficult to get the sensor positioned correctly.

Can someone please explain how this is suppose to work as I have spent days googling, reading and going through the source code but am still confused as to how this should work.
[Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/2391660/Configuration.Files.zip)

I am running the latest 1.1 version on git.

forkoz, 2018-09-19:
Does it actually print at .3? Because after homing the nozzle is not at Z0. I always send G1 F60 Z0 when setting the offset to be sure I'm at 0. You don't want your nozzle hot and sitting on the bed after a home.

Also if you set negative z offset the nozzle is below sensor trigger point by that amount @ Z0. Positive offset = above the bed.

totalretribution, 2018-09-19:
I just ran a M502 and checked that both M851 and M206 are set to 0 (added to bottom of this post). I then homed and measured the nozzle to bed and it measured as 0.3.

I then set M851 to Z0.3 and homed, the nozzle measured as 0.3 and the co-ordinate states -0.3 which is incorrect.

I then set M851 to Z-0.3 and homed, the nozzle measured as 0.6mm from the bed and the co-ordinate states 0.3. I then performed a G1 Z1 and G1 Z0 and measured the nozzle again which was 0.6mm. During this home I noticed that the Z axis performs an extra move right at the end.

I did all the measurements with a feeler gauge. I also measured the inductive sensor to bed as 0.75mm when the nozzle is 0.3mm from the bed.

What I don't understand is why M851 with a negative value makes the Z axis move by the value set and then set the co-ordinates for Z to that set value. Is this what its meant to do?

`M502
M503
echo:  G21    ; (mm)
echo:  M149 C ; Units in Celsius

echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E93.00
echo:Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X500 Y500 Z100 E5000
echo:Acceleration (units/s2): P<print_accel> R T
echo:  M204 P500.00 R500.00 T500.00
echo:Advanced: B S T X Y Z E
echo:  M205 B20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Unified Bed Leveling:
echo:  M420 S0 Z0.00

Unified Bed Leveling System v1.01 inactive.

Active Mesh Slot: -1
EEPROM can hold 7 meshes.

echo:Material heatup parameters:
echo:  M145 S0 H190 B60 F255
echo:  M145 S1 H240 B100 F255
echo:PID settings:
echo:  M301 P21.73 I1.54 D76.55
echo:Z-Probe Offset (mm):
echo:  M851 Z0.00`

thinkyhead, 2018-09-19:
> I don't understand is why M851 with a negative value makes the Z axis move by the value set and then set the co-ordinates for Z to that set value. Is this what its meant to do?

I don't understand it either. The raise of Z at the end of `G28` is not normal.

After homing the Z position is supposed to be your `M851 Z` value negated. For example, if your nozzle measures 0.3mm off the bed after the probe triggers (for example, immediately after a `G28`), the probe Z offset should be set as `M851 Z-0.3`. This will cause the current Z position to be 0.3 after homing Z. A move with `G1 Z0` then puts the nozzle right up to the bed.

You shouldn't do anything with `M206`. It's not applicable, so just leave your home offsets at zero.

To get to the bottom of this, please do the following:

- Download Marlin `bugfix-1.1.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Copy the log output into a **.TXT file** and attach it to your next reply.

From the log we should be able to see what is happening during `G28` to produce the unusual result you see.

totalretribution, 2018-09-19:
@thinkyhead 
Thanks for the clarification, M851 works how I originally expected. I did not think the raising of Z was correct but I was not 100% sure.

I have just done a new build using a fresh portable arduino, clean download of marlin and instead of copying the config files I started from the ender 3 examples and edited them (i use a mks_gen_l)  and it does not seem to be doing it now.

I will experiment for a bit and update this when I know more.

forkoz, 2018-09-20:
Well I just did homing and my Z after G28 is +1.47, my z offset is -1.47.

totalretribution, 2018-09-26:
Im unable to reproduce this and I used beyondCompare to compare the builds and I could not find any real difference.

closed 2018-09-26 by totalretribution