MarlinFirmware/Marlin #11870: TMC2130 only works right after flashing. After reset no more go.

hapklaar, 2018-09-18:
I have the strangest issue with the new 1.1.9 release, tailored for my AnyCubic Kossel with 3x tmc2130.

After flashing the unit works just fine. No skipping in this fw as far as I've been able to test and wonderfully silent due to the tmc's.

But after a reboot the unit does not drive the X, Y and Z motors anymore. It just sits there doing nothing after a G28 or any other g code which moves any of these motors. Only the extruder motor remains functional, which is still driven by an A4988.

To get it operational again I need to re flash the exact same fw code, after which it works until it is restarted again.

I don't have this issue when using all A4988. Not sure where to start debugging this...

thinkyhead, 2018-09-19:
New drivers might help. File a new report if there's still an issue. I'm not sure if anyone has gotten TMC drivers to work with a Trigorilla board, but search the issue queue to see if there's some mention of both together.

Have you wired up the drivers for SPI communication? Are you using really good heatsinks and active cooling for the drivers? It can be easy to accidentally fry them if you just plug them in and try to use them without first tuning and cooling them.

closed 2018-09-19 by thinkyhead

hapklaar, 2018-09-19:
Yeah, it should be able to work according to https://www.youtube.com/watch?v=GF6ZGKOuzhY . I also use the same configuration.

I used https://cdn.thingiverse.com/assets/b6/a9/0d/2d/d1/tmc2130_on_Anycubic_kossel_trigorilla.pdf for SPI wiring diagram.

Will report back when I have new drivers.

hapklaar, 2018-09-24:
After reviewing my SPI configuration, I discovered I forgot to assign X_CS_PIN/Y_CS_PIN/Z_CS_PIN to the right board pins. After changing this in ramps.h and re flashing, I had no more issues (still on Chinese clones)

thinkyhead, 2018-09-28:
You should assign/override the CS pins in your `Configuration.h` file so they won't get lost in future updates.