MarlinFirmware/Marlin #11871: Bugfix-2.0.x Pronterface upload to SD does not work

gloomyandy, 2018-09-18:
If you attempt to upload a file to the SD card via Pronterface with the current version of Marlin the upload will not complete. Instead you will get one or more errors of the form:
[ERROR] Error:No Checksum with line number, Last Line: nnnn
and finally one of the form:
[ERROR] Error:No Checksum with line number, Last Line: -1
and all subsequent commands will be ignored. This problem seems to have been introduced by the following change:
https://github.com/gloomyandy/Marlin/commit/3007eee22e837ad45c25213dfba4f706f0327e4a#diff-36487460db011f5cf1bfec0ab62b91bd
which basically requires that all lines sent to Marlin following the start of an upload must have a line number and checksum. Unfortunately Pronterface sends the M29 command (to indicate the end of the upload) without the linenumber and checksum. If the Pronterface option to "monitor printer status" is enabled Pronterface will also periodically send M105 commands during the file upload again these will not have a line number and checksum and generate the first set of errors above.

Note that with the code before the above change although the upload appears to work it seems that the periodic M105 commands ended up being stored in the SD card file, which is probably not correct. 

Reading the following:
https://github.com/MarlinFirmware/Marlin/pull/10204
it seems that it was known that this was an issue with Octoprint but it was possible to work around this by setting an Octoprint option to always use line numbers and checksums. Is there a similar option for Pronterface I couldn't find one?

Arguably this is a Pronterface issue, but I couldn't find a report of the issue here, so thought it was worth reporting.

svs072477, 2018-09-19:
Try commenting  these lines in the file queue.cpp
/*        #if ENABLED(SDSUPPORT)
          else if (card.saving)
            return gcode_line_error(PSTR(MSG_ERR_NO_CHECKSUM), i);
        #endif */

gloomyandy, 2018-09-19:
Yep those changes will definitely let things "work" (I made this to test out the old config), but you do end up with spurious M105 commands in your SD card file, which is probably not a problem, but not ideal.

svs072477, 2018-09-19:
I agree, this bug need to fix

p3p, 2018-09-19:
As the reprap gcode standard states, when transferring a file with M28 - M29 all commands between those commands are stored on the sdcard, so in this case your host software is not obeying the standard by sending those M105 commands, in theory we could strip them but then it would be us not obeying the standard, and there will always be someone that wanted those commands in their file we just changed in transit.

thinkyhead, 2018-09-19:
Our solution will be to look for `M29` and treat it as a special case.

thinkyhead, 2018-09-27:
The `M29` being sent by the host is breaking the rule that **a checksum is required when there is a line number**. If the command didn't include a line number then it wouldn't be required to have a checksum.

So, the question is, why is host software sending `M29` with a line number, but not a checksum?

A good question for the @MarlinFirmware/host-software-team

thinkyhead, 2018-09-27:
The issue seems to be:
```cpp
        if (npos) {
          . . .
        }
        #if ENABLED(SDSUPPORT)
          else if (card.saving)
            return gcode_line_error(PSTR(MSG_ERR_NO_CHECKSUM), i);
        #endif
```
I'm not sure why, but if there's no line number, then we're throwing this (misleading) error when `card.saving`. It might be correct to pass on all commands that lack a line number, but evidently `M29` is an exception.

I'm still curious as to why host software sends the `M29` without a line number. And evidently we do need to pass the `M29` through even though it fails the test. So, this would be the place to add an exception for `M29`.

CC: @MarlinFirmware/host-software-team

thinkyhead, 2018-09-27:
Merged a workaround, so it won't throw an error for `M29` when `card.saving`.

boelle, 2019-02-21:
@gloomyandy is problem solved with latest bugfix 2.0?

TVAVAE, 2019-03-03:
> @gloomyandy is problem solved with latest bugfix 2.0?

No, not when I use it.

gloomyandy, 2019-03-03:
A "fix" was applied but I do not believe it actually works. See this PR thread: https://github.com/MarlinFirmware/Marlin/pull/13013

This thread also includes two proposed solutions, neither of which have (as far as I know) been applied.

gloomyandy, 2019-03-03:
Perhaps @thinkyhead could review the PR?

boelle, 2019-03-13:
@thinkyhead thoughts?

TVAVAE, 2019-04-13:
Having the same problem here with a own board. Both octopi and pronterface have problems with writing files to the SD at different baudrates.

Example problem:
Resend: 11460
Resend: 12603
Error:Line Number is not Last Line Number+1, Last Line: 12603
[ERROR] Error:Line Number is not Last Line Number+1, Last Line: 12603

Resend: 12604
Error:No Checksum with line number, Last Line: -1
[ERROR] Error:No Checksum with line number, Last Line: -1

Resend: 0
Resend: 0
...

Keep me updated! STM32F4 here.

MuellerDominik, 2019-04-21:
In my opinion this issue can be closed.

To summarize, it is now possible to send a `M29` command with a line number and checksum e.g. `N4 M29*28` or without a line number and no checksum i.e. `M29`. Both will properly be accepted and stop writing to the file, which was the initial problem described in this issue. This is due to the following lines in the different branches:

* `1.1.x` branch: https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L1094
* `bugfix-1.1.x` branch: https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Marlin_main.cpp#L1089
* `bugfix-2.0.x` branch: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/queue.cpp#L630

There was a second (similar) bug which probably led to some confusion. In the `bugfix-2.0.x` branch it wasn't possible to stop writing to a file, if the `M29` command was sent with a line number and checksum e.g. `N4 M29*28` (since it was only checking, if index 0 of the command is equal to the char `'M'`, index 1 equals `'2'`, ...). It wasn't properly looking for a substring. This has been fixed 5 days ago in the following commit: https://github.com/MarlinFirmware/Marlin/commit/92ed9a2bb6d90a969f2ceaf0a73ebe3c0cdc4c16.

@TVAVAE, I believe this should have fixed your issue. Try using the latest version of the `bugfix-2.0.x` branch (BTW. I like to hear that other people are using SMT32 microcontrollers, ATM. we're designing a new board which uses the STM32F103ZE).

There is one more bug in the `1.1.x` and `bugfix-1.1.x` branches. The parser is not properly checking, if the character following the string `M29` is numeric, when the substring `M29` is found in the command. This means that the command `M290` (which is [available since v1.1.7](http://marlinfw.org/docs/gcode/M290.html)) will wrongly stop the SD transfer. I have already opened an issue: https://github.com/MarlinFirmware/Marlin/issues/13777.

I just hope no one is using the string `M29` in a G-code comment i.e. `; M29 stops writing to a file` (if comments don't get stripped before serial SD upload).

closed 2019-05-16 by thinkyhead