MarlinFirmware/Marlin #11873: [2.0.x Feature request] G-code exchange and transfer files to sd card from devices via SPI+DMA

svs072477, 2018-09-19:
Need a new feature in Marlin 2.0. 
There is an EPS32 with a web interface to control the 3D printer and write files to the sd card (sd card reader on RepRapDiscount Display), now the connection between the ESP32 and the printer Board is via UART, while recording to the sd card is very slow. 
If connect ESP32 to the printer Board via SPI with DMA, recording will be much faster. 
SPI in ESP32 can work in master and slave modes. 
Is it possible to add to Marlin the ability to exchange commands and transfer files from the SPI device?
Most likely will need to use a separate SPI interface that is not associated with SD card, TMC2130 and other devices

gloomyandy, 2018-09-19:
Can you provide links to the EPS32 device that supports this feature? The only one I'm aware of requires additional hardware on the control board to allow switching of the SPI bus between the EPS32 device and the control board.

ejtagle, 2018-09-19:
The only problem i see is that the STM32 must be able to handle that SPI communication at such speeds. Even if using DMA it could be possible to receive the data, a different thing is if we will be able to process it...

svs072477, 2018-09-19:
Well for a start I can connect ESP32 and STM32F429, to STM32 to connect SD CARD on SDIO, and try to write the file from C ESP32 to the card connected to STM32. The exchange of commands can be left on the UART, I'm more concerned about recording on the map via wifi.
I think that I can handle, although not particularly good at programming.

boelle, 2019-02-21:
@thinkyhead maybe add the feature request label?

kresike, 2019-08-13:
> Well for a start I can connect ESP32 and STM32F429, to STM32 to connect SD CARD on SDIO, and try to write the file from C ESP32 to the card connected to STM32. The exchange of commands can be left [...]

I would like to help with this, have some programming skills. Did you succeed in connecting the ESP and the STM ? If so, could you share a schematic ?

sjasonsmith, 2020-09-11:
@svs072477 Marlin 2.0 has an experimental `BINARY_FILE_TRANSFER` which can be enabled in `Configuration_adv.h`. Multiple people have reported it works well, and there is even an [OctoPrint plugin](https://plugins.octoprint.org/plugins/marlinbft/) which implements it.

Does that like it would meet your needs?

ellensp, 2026-11-04:
Marlin has moved on a lot since this was opened, so here's where this stands.

**What SPI between an ESP32 and the controller would take**

- It needs custom firmware on the ESP32, since nothing that talks to Marlin today (ESP3D, the MKS WiFi firmware) uses this link.
- Marlin would need to receive commands and files as an SPI slave. Nothing in Marlin does that today. The STM32F1 HAL's SPI class has a `beginSlave()`, but nothing uses it, so this would be new code. It could start with one MCU family and grow from there.
- It needs SPI between the two chips. Some boards have SPI pins on their WiFi header, such as the BTT SKR V2 and Octopus, but Marlin only uses UART there, and other boards would need a custom board.
- It needs a new protocol, with framing and error recovery, for both commands and files.

For comparison, RepRapFirmware uses this design, with its own ESP firmware, SPI slave code with DMA, and its own protocol.

A faster link also doesn't change the other limit: Marlin still writes to the SD card from its main loop while it runs the printer.

**What's available now for faster file transfer**

- **ESP32 boards:** Marlin runs on the ESP32 itself, so there's no link between two chips. With `ESP3D_WIFISUPPORT`, files can be uploaded to the SD card directly over WiFi.
- **STM32 boards with an MKS WiFi module:** `MKS_WIFI_MODULE` (under `WIFISUPPORT`) gives you what this issue asks for without SPI. The link normally runs at 115200 baud, but for a file upload it switches to 1958400 baud with DMA, then switches back. That's about 8 times faster than 250000 baud, over the same two UART wires.
- **Any board with a host connected:** `BINARY_FILE_TRANSFER` in `Configuration_adv.h`, as suggested above, along with a higher `BAUDRATE` or native USB where the board has it.

If one of these doesn't cover your use, please reply with the board you're using and what you measured, since that would make a request for a specific faster link something we can act on.