MarlinFirmware/Marlin #11874: when starting to print printer goes negative X and hangs up

edism, 2018-09-19:
I have compiled the latest version 1.1.9 and uploaded to my printer. I use similar style as CR-10 with MKS gen 1.4 board.

All works well, printer is auto homing, all axes, temperature control and everything else seems to work well.

But if I print a file from SD card (haven't tried printing not from SD) printer heats up, homes all axis and then suddenly X axis start to go negative - carriage jams into the endstop, stepper starts to skip steps from extreme load and interface hangs up ... no controls work on LCD scree and then after sometime Marlin reboots.

Any ideas what's wrong here? Not sure if config files could mess it up in such way.

edism, 2018-09-20:
I've been using Marlin for few years now and did multiple upgrades, but this time I'm not sure what's wrong and where to look for a problem.

SD file is ok. I used the same files i've printed on older version, plus i have made couple new ones to double check this idea. I need to use SD check and retry option, which is a bit out of standard config.

From LCD menu all axes work fine, move to right directions.  Auto home feature also works correct. Something happens after homing to 0 when trying to print.
Please see attached gcode file.

[pzzle part - no sprt.txt](https://github.com/MarlinFirmware/Marlin/files/2399443/pzzle.part.-.no.sprt.txt)

edism, 2018-09-25:
Could anyone provide any thoughts how to troubleshoot this? I have went over my config settings again an d can't find anything that could cause this...

bryanmee, 2018-09-26:
did you enable - #define ENDSTOPS_ALWAYS_ON_DEFAULT
but not -   #define ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED

edism, 2018-09-27:
both of these options are disabled in my config.

> // If you want endstops to stay on (by default) even when not homing
// enable this option. Override at any time with M120, M121.
//#define ENDSTOPS_ALWAYS_ON_DEFAULT

> 

>   /**
   * This option allows you to abort SD printing when any endstop is triggered.
   * This feature must be enabled with "M540 S1" or from the LCD menu.
   * To have any effect, endstops must be enabled during SD printing.
   */
  //#define ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED

> 

>

edism, 2018-09-27:
Please see attached my config files, if that would help...
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2422292/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/2422293/Configuration_adv.txt)

bryanmee, 2018-09-27:
The two main difference between your settings and the ones I use are:

Most of the time the probe is lower than the nozze or it would not be able to touch the bed.  So a negative number is used.  Your's is positive.
#define Z_PROBE_OFFSET_FROM_EXTRUDER 1   // Z offset: -below +above  [the nozzle]

#define NO_WORKSPACE_OFFSETS - I am not using this.

edism, 2018-09-27:
This is true, I use inductive sensor and a 2mm glass, so my sensor is actually 1 mm above the nozzle, this worked with previous version no problem..

As for the NO_WORKSPACE_OFFSETS i will give a try without this and see if this makes any difference...

Just the behavior is odd. printer freezes all 3 axes and starts rocking extruder back and forth, like thinking that X.Y.Z moves are instant and progress bar goes 50 times quicker

AnHardt, 2018-09-27:
Tha config does not deal with the shell of your sensor, but with the invisible trigger point (depending on the exact type of the sensor 2, 4, 8 mm) below its surface.

edism, 2018-09-27:
So can this setting cause this odd behavior?

bryanmee, 2018-09-27:
I had a problem were an endstop was triggered when I was printing.  This causes them command queue processor to abort the current command and remove it from the queue.  The Marlin thought the command had been completed so the position was off by the amount of the lost command.  This happened multiple time until the steppers run into the endstops (making bad noises).  When I looked at the Octoprint log there was “endstop Hit:”  You could rule this out by issue by disabling bed leveling  or setting your probe offset very low and rebuild table
#define Z_PROBE_OFFSET_FROM_EXTRUDER -5 // Z offset: -below +above [the nozzle]

thinkyhead, 2018-09-28:
Try this… Start by making a G-code file that only includes the start G-code (as shown in your G-code file) plus a move of XY to the middle of the bed. Save it as `test.gcode`.
```gcode
G90
M82
M106 S0
M104 S210 T0
M109 S210 T0
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M111 S247 ;debugging
G28 ; Home extruder
M114 D ;detailed position
G1 Z15.0 ;move the platform down 15mm
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G92 E0
G1 E-2.9000 F1800
G1 Z0.200 F1800
T0
M114 D
G1 X100 Y100 F3600
G1 E0.0000 F540
```

Put this test file on your SD card. Then set up your printer with maximum debugging:

- Download Marlin `bugfix-1.1.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Clear the host console.
- Start the `test.gcode` file on the SD card.

Copy the console output into a `log.txt` file, save it, and include this file in your next reply.

edism, 2018-10-03:
I managed to get it to work by leaving all default settings from last version and only changing my endstop and steps/mm settings. Something in the config makes it to behave this way...

thinkyhead, 2018-10-03:
If you only change one option at a time between flashes, then hopefully you'll catch the one that causes the issue. Keep us posted!

PeterGrace, 2018-10-11:
I am seeing a similar problem on Marlin 1.1.9.  I added some script commands to try to help diagnose what's going on, and if I do an M114 right before the crash,  it says that it is at Z0, but the Count is negative and large:

```
Recv: echo:busy: processing
Recv: ok N14 P15 B3
Send: N15 M114*19
Recv: X:293.00 Y:258.00 Z:0.00 E:0.00 Count X:23440 Y:20640 Z:-2147483648
Recv: ok N15 P15 B3
Send: N16 G1 Z15 F300*52
Recv: ok N16 P14 B3
Send: N17 G4 S5*93
```

I'm not sure where this Count value comes from, but when I look at the LCD, when this happens, the Z field fills with upper-ascii characters instead of a number, leading me to think that maybe there's some memory corruption issues involved.  I do have a very tight progmem binary.

edism, 2018-10-16:
Have you tried getting a clean version and configuring a bare minimum that needs to be configured to make the machine work?

thinkyhead, 2018-10-16:
Interestingly, -2147483648 = 0xFFFFFFFF80000000

PeterGrace, 2018-10-16:
I recompiled Marlin without sdsupport (as I don't use it), and that saved a fair bit of memory.  The issue went away.

thinkyhead, 2018-10-26:
It's a conundrum. Maybe there's a buffer overrun someplace. Maybe something else. Chances are good the issue will get fixed in the course of general patching of the 2.0.x code-base. Since 1.1.x is now end-of-life, we won't be pursuing the issue with that version.

closed 2018-10-26 by thinkyhead