MarlinFirmware/Marlin #11876: Question - wait_for_heating versus M109 method

xC0000005, 2018-09-19:
While looking at a question about PID instability when the hotend has cooled down during an M600 filament change, I noticed something that might be off. M109 has a nice loop where it requires the temperature to reach a given range and hold within that for a given time. The M600 resume code uses (eventually) wait_for_heating, which returns true once the temperature is within the degree range at all. 

Should the pause resume code use the same logic as M109? A hotend heating up from a cold temp often has an overshoot, followed by a drop below target temp, and then levels out.

GMagician, 2018-09-20:
> Should the pause resume code use the same logic as M109?

Well since heating is stopped and then reactivated I think it should be stable before resuming print so, in my opinion, the answer is yes

thinkyhead, 2018-09-21:
I agree that we should use the same method. If no one else submits the code I’ll do it early this week.

GMagician, 2018-09-21:
I checked code and I tryed to collect common code to be used for m600 and m109 (taking a look also at M104) but I see something I don't understand (and not related to this fix).
planner AUTOTEMP method is always called by M104 but not always in M109...is it right?

xC0000005, 2018-09-21:
I am traveling for the next couple weeks, any code changes I do will be theoretical since I can't test them. So, for instance, I'm working on the next iteration of my unified STM32 hal (based pretty much on the STM32F4 one currently in), I won't be able to actually test beyond compiling any change I do. Sorry. If it can wait, I'll refactor it when I get back.

thinkyhead, 2018-09-28:
> planner AUTOTEMP method is always called by M104 but not always in M109...is it right?

I just looked at the code and both `M104` and `M109` call `planner.autotemp_M104_M109()`.

GMagician, 2018-09-29:
> I just looked at the code and both M104 and M109 call planner.autotemp_M104_M109().

Not exactly "always".
M104:
```cpp
if (parser.seenval('S')) {
      [cut]
      }
    #endif
  }

  #if ENABLED(AUTOTEMP)
    planner.autotemp_M104_M109();
  #endif
```

M109:
```cpp
  if (no_wait_for_cooling || parser.seenval('R')) {
    [cut]
  }
  else return;

  #if ENABLED(AUTOTEMP)
    planner.autotemp_M104_M109();
  #endif
```

as you can see there is a "return"

thinkyhead, 2018-09-29:
I see. If `M109` is called without an `R` or `S` parameter, it returns because it's considered an error. But with `M104` it does always call it, even if no `R` or `S` is given.

Seems as though `M109` should also call it if there's an `F` parameter.

thinkyhead, 2018-09-29:
Ok, now patched.

GMagician, 2018-10-02:
I think this may be closed. posted PR seems to work

closed 2018-10-03 by thinkyhead