MarlinFirmware/Marlin #11878: Marlin 1.1.9 Auto home G28 failsafe

Pyriame, 2018-09-20:
### Description
hi,
I install Marlin 1.1.9 on a CR-10 S4 (400mmx400mmX400mm). 
I run a G28 (auto home) X moves until it stops and Y moves about 10cm and the printer makes a safe. I turn off and turn on the printer. I launch again G28. X moves up to the stop and Y moves about 10cm further and then again safe. I stubborn and I turn back on. New G28 and this time Y moves, meets it stop, Z goes down and the auto home is OK.

Same problem at the end of an impression

### Steps to Reproduce

just launch G28 with big Bed 40cmx40cm 

#### Additional Information

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2399089/Configuration.zip)

AnHardt, 2018-09-20:
Denoise your Y-endstop.

Pyriame, 2018-09-20:
my bed does not have time to move to y-endstop
is there an option in Marlin to define a maximum or minimum displacement length that generates failsafe?
Normally there is a displacement that the truck does not meet the Endstop? With the original firmware I had never had this problem

thinkyhead, 2018-09-20:
When homing a move is done for double the defined width or depth. If the procedure is stopping early, there's something else unusual going on. Try enabling `ENDSTOP_NOISE_FILTER` to see if it helps.