MarlinFirmware/Marlin #11881: BUG with //#define EEPROM_SETTINGS. If the option is activated, the x motors, and it works poorly

oriaj3, 2018-09-20:
### Description

BUG with //#define EEPROM_SETTINGS.
If the option is activated, the x motors, and it works poorly

### Steps to Reproduce

1. MKS Sbase v1.3, TMC2208 and CNCSHIELD. 
2. Activate in configuration.h
"#define EEPROM_SETTINGS."

3. The motors of (x, y) go to blows and stop.

**Expected behavior:** Everything works correctly.

**Actual behavior:** The motors of (x, y) go to blows and stop.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-09-20:
Have you done `M502`, `M500` after flashing the firmware to initialize the EEPROM?

oriaj3, 2018-09-20:
No, I have not.

oriaj3, 2018-09-20:
> Have you done `M502`, `M500` after flashing the firmware to initialize the EEPROM?

Thank you very much that solved the problem. @thinkyhead

closed 2018-09-20 by oriaj3