MarlinFirmware/Marlin #11882: Z axis issue

seaquest37, 2018-09-20:
### Introduction
Past 3 years, I've bought a cheap Migbot i3 3D printer and it's find for my needs.  The electronics was a board MKS Base and, I've upgrate it to a new MKS Base v1.6 and it's work fine until 2 past days with the Marlin release 1.1.6.

For a reason that I dont know, my board stuck and stop answer.  Having lost my previous 1.1.6 Configuration.h (Windows 10 Update crash issue), so I've restart from fresh, and everything work well except for the Z Axis.

### Description
The Z axis seems to run too fast or are out of sync.  To made a quick test, I've swapped my Z axis with the X axis, request a Z move about 0.1mm and the X axis moved about 5mm.

I request a 0.1mm move ont the X axis, the Z axis moved for 0.1mm.  The the issue doesn't seems to be the servos.

The step motors model is Busheng 17HD40005 22B

In the Configuration.h, I've found this:

/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 500 }

So if I understand correctly, the default Steps/mm for the Z Axis is set to 4000?  It's not a bit to high regarding the X and Y axis?

Any help will be appreciated

Thank you,

Martin

seaquest37, 2018-09-20:
Ok, i've found this explanation into this web page: 
[3D Printer Firmware Settings - Stepper Motor Configuration](https://www.matterhackers.com/news/3d-printer-firmware-settings-stepper-motor-configuration)

`#define DEFAULT_AXIS_STEPS_PER_UNIT {AXIS_STEPS_PER_UNIT_X, AXIS_STEPS_PER_UNIT_Y, AXIS_STEPS_PER_UNIT_Z, AXIS_STEPS_PER_UNIT_E}`

> {snip}...Each of these values is calculated separately.  For belt driven axes (usually x and y), the steps per unit is determined by the number of steps per revolution divided by the idler teeth, divided by the belt pitch (ie the spacing between teeth):

> Steps per Unit (X and Y Axes) = Motor Steps per Revolution / Idler Teeth / Belt Pitch

> For threaded rod driven axes, steps per unit is calculated as steps per revolution divided by the pitch of the rod:

> Steps per Unit (Z Axis) = Motor Steps per Revolution / Rod Pitch

> For geared extruders, the calculation is steps per revolution times the gear ratio divided by the pinch wheel diameter times pi:

>Steps per Unit (Extruder) = Motor Steps per Revolution * Extruder Gear Ratio / (Pinch Wheel Diameter * Pi)

A lot's of new words... I guess it's a language limitation, but idlter Teeth, belt pitch, rod pitch... somebody willing to give me mecanical 101 crash course or documentation?

Thanks

AnHardt, 2018-09-20:
https://www.prusaprinters.org/calculator/

thinkyhead, 2018-10-08:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube

closed 2018-10-08 by thinkyhead