MarlinFirmware/Marlin #11883: [Bugfix-2.0.x] Error:EEPROM datasize error.

Ludy87, 2018-09-20:
### Description

I flash Marlin new for update from 1.1.9 to Bugfix-2.0.x and i see EEPROM ERROR.

Result from `#define DEBUG_EEPROM_READWRITE`

	echo:Marlin bugfix-2.0.x
	echo: Last Updated: 2018-01-20 | Author: (none, default config)
	echo:Compiled: Sep 20 2018
	echo: Free Memory: 2675  PlannerBufferBytes: 1472
	Error:Field lcd_preheat_hotend_temp mismatch.
	[ERROR] Error:Field lcd_preheat_hotend_temp mismatch.

	Error:EEPROM datasize error.
	[ERROR] Error:EEPROM datasize error.

	echo:Index: 581 Size: 593
	echo:Hardcoded Default Settings Loaded

Ludy87, 2018-09-20:
i tried and the Result is be same

seaquest37, 2018-09-20:
Hi Ludy87,

I suggest to fix the first error.

Error:Field lcd_preheat_hotend_temp mismatch.
[ERROR] Error:Field lcd_preheat_hotend_temp mismatch.

In the Marlin_main.cpp, we found: 
lcd_preheat_hotend_temp[material] = constrain(v, EXTRUDE_MINTEMP, HEATER_0_MAXTEMP - 15);

Check in the Configuration.h if you have a kind of typo for theses variables like:

EXTRUDE_MINTEMP = 270,5; //instead of 270.5 or 270.; etc..
HEATER_0_MAXTEMP 

I guess this caused the "mismatch" and change the expected size value of the compiled EEPROM.

Let us know.

cjsoong, 2018-09-20:
@Ludy87 
Do you use z probe servo?
Disable these four items try again! (need to do M502 & M500)
//#define Z_PROBE_SERVO_NR 0
//#define Z_SERVO_ANGLES {170,111}
//#define NUM_SERVOS 1
//#define EDITABLE_SERVO_ANGLES

Ludy87, 2018-09-20:
@seaquest37 No, nothing like that found

@cjsoong this is my problem. Whats is the alternative?

cjsoong, 2018-09-21:
@Ludy87 
Try my modified version.
I am using it on my machine now.
https://drive.google.com/file/d/1wEF-51bp7OMicY-_3ljFHX5rOZP5Mip2/view?usp=sharing

Ludy87, 2018-09-21:
Thank you @cjsoong i have this code changed 

    // AUTO_BED_LEVELING_UBL
    //
    bool planner_leveling_active;                         // M420 S  planner.leveling_active
    int8_t ubl_storage_slot;                              // ubl.storage_slot
    
    //
    // SERVO_ANGLES
    //
    //uint16_t servo_angles[NUM_SERVO_PLUGS][2];                 // M281 P L U
    uint16_t servo_angles[NUM_SERVOS][2];                 // M281 P L U

    //
    // DELTA / [XYZ]_DUAL_ENDSTOPS
    //

---
@thinkyhead can it to complications come in?

alexxy, 2018-09-23:
Seems like same issiu as #11757

Ludy87, 2018-10-06:
https://github.com/MarlinFirmware/Marlin/commit/3041f06ad1f00b1895ccedb3a3dcbc32fc01cff9