MarlinFirmware/Marlin #11885: Layer misalignment

14015, 2018-09-20:
Hello,
I'm using cartesian 3D printer, which is my own project, with Marlin firmware and I have the following problem:
When I print something larger, in first 3 milimeters the print overhangs (it is like layer misalignment) and after the 3rd milimeter it is stable. 
It seems like layer misalignment or steps skipping, but only for the first 3 milimeters, only in X AXIS and only in one direction EVERYTIME on larger prints. 
I've tried with different filaments, different temperatures and different speeds and the problem is the same everytime. 
Do you have any idea how can I fix this? Is it something like scew correction in Marlin firmware but only for the first 3 milimeters ?

P.S. The printer is with linear movement and I'm shure that the problem is not in the belts. 

Thank you! :)

[image]
[image]

dgrat, 2018-09-21:
I would guess that based on your pictures there are several problems. 

- Your extrusion rate seems too high
- Layer shifts are commly caused by too high or too low current on the stepper driver, or if there is too much friction on one of the axes
=> I sometimes touch the driver, if it feels like you burn yourself in the moment you touch, it is probably too much current, if it feels just hot like 90/100°C it might be fine.
- I have the feel in your case one axis might be lose, because in my first printer I had a similar looking problem once (it was the z axis connector)

14015, 2018-09-21:
Thank you for your response! 
Now everything works fine. 
I'm using Nema23 with PoStep256 drivers and they operate on its normal temperature - around 50°C, but I lowered the current because the stepper doesnt sounds well (on lower speeds) . 
The stepper motor was 30°C and may be it doesn't works well on lower temps. 
I changed the current only on this axis - from 2 to 3Amps and now the stepper motor is on 70°C which is normal. 
The problem with the lower print quality is expected, because I'm using 1mm nozzle and 400mms print speed. 
I changed only the current and nothing else, but the impact is huge. 
Thanks a lot! 
[image]
[image]

closed 2018-09-21 by 14015