MarlinFirmware/Marlin #11888: [bugfix-2.0.x] Won't compile with vscode

wsidebottom, 2018-09-21:
### Description
Won't compile with windows + vscode + platformio
Compiles correctly on ubuntu 16.04 using platformio doing same steps

### Steps to Reproduce
1. Download repo and switch to bugfix-2.0.x branch
2. Change platformio.ini to build LPC1768
3. Change Configuration.h & Configuration_adv.h to examples in MKS/SBASE directory
4. Change Configuration.h to define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
5. Execute 'platformio run -v'

#### Additional Information
Output from windows:
PS D:\Source\Marlin> platformio run -v

Processing LPC1768 (board_upload.maximum_ram_size: 32768; board_build.f_cpu: 100000000L; build_flags: !python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py -fmax-errors=5, -g, -ggdb; lib_deps: CMSIS-LPC1768, https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip, TMC2130Stepper@>=2.2.1; debug_tool: custom; build_unflags: -Wall; extra_scripts: Marlin/src/HAL/HAL_LPC1768/debug_extra_script.py, Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py, Marlin/src/HAL/HAL_LPC1768/upload_extra_script.py; platform: nxplpc; debug_server: C:\Program Files (x86)\SEGGER\JLink_V618d\JLinkGDBServerCL.exe, -select USB, -port 2331, -device LPC1768, -if JTAG, -speed auto, -noir; board: lpc1768; lib_extra_dirs: frameworks; monitor_speed: 250000; lib_ldf_mode: off; src_filter: +<src/*> -<src/config>)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Storage: D:\Source\Marlin\.piolibdeps
Looking for CMSIS-LPC1768 library in registry
Warning! Library `{'requirements': None, 'name': 'CMSIS-LPC1768'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'CMSIS-LPC1768'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
LibraryManager: Installing dev
U8glib-HAL @ 0.4 is already installed
LibraryManager: Installing id=1493 @ >=2.2.1
TMC2130Stepper @ 2.5.0 is already installed
PLATFORM: NXP LPC > NXP mbed LPC1768
SYSTEM: LPC1768 100MHz 32KB RAM (512KB Flash)
DEBUG: CURRENT(custom) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(soft)
Collected 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CMSIS-LPC1768> 0.0.0 (D:\Source\Marlin\frameworks\CMSIS)
|-- <U8glib-HAL> 0.4 [https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip] (D:\Source\Marlin\.piolibdeps\U8glib-HAL_ID1932)
|-- <TMC2130Stepper> 2.5.0 (D:\Source\Marlin\.piolibdeps\TMC2130Stepper_ID1493)

Unable to find destination disk.  File must be copied manually.

arm-none-eabi-g++ -o .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Os -ffunction-sections -fdata-sections -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m3 -mthumb -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing -fsingle-precision-constant --specs=nano.specs --specs=nosys.specs -MMD -MP -fmax-errors=5 -g -ggdb -DF_CPU=100000000L -DPLATFORMIO=30601 -DU8G_HAL_LINKS -DTARGET_LPC1768 -IMarlin -Iframeworks\CMSIS\LPC1768\include -Iframeworks\CMSIS\LPC1768\lib -Iframeworks\CMSIS -I.piolibdeps\U8glib-HAL_ID1932\src -I.piolibdeps\TMC2130Stepper_ID1493\src -IMarlin\src\HAL\HAL_LPC1768\include -IMarlin\src\HAL\HAL_LPC1768\u8g Marlin\src\HAL\HAL_LPC1768\HAL.cpp
arm-none-eabi-g++ -o .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_spi.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Os -ffunction-sections -fdata-sections -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m3 -mthumb -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing -fsingle-precision-constant --specs=nano.specs --specs=nosys.specs -MMD -MP -fmax-errors=5 -g -ggdb -DF_CPU=100000000L -DPLATFORMIO=30601 -DU8G_HAL_LINKS -DTARGET_LPC1768 -IMarlin -Iframeworks\CMSIS\LPC1768\include -Iframeworks\CMSIS\LPC1768\lib -Iframeworks\CMSIS -I.piolibdeps\U8glib-HAL_ID1932\src -I.piolibdeps\TMC2130Stepper_ID1493\src
-IMarlin\src\HAL\HAL_LPC1768\include -IMarlin\src\HAL\HAL_LPC1768\u8g Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp
arm-none-eabi-g++ -o .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Os -ffunction-sections -fdata-sections -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m3 -mthumb -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing -fsingle-precision-constant --specs=nano.specs --specs=nosys.specs -MMD -MP -fmax-errors=5 -g -ggdb -DF_CPU=100000000L -DPLATFORMIO=30601 -DU8G_HAL_LINKS -DTARGET_LPC1768 -IMarlin -Iframeworks\CMSIS\LPC1768\include -Iframeworks\CMSIS\LPC1768\lib -Iframeworks\CMSIS -I.piolibdeps\U8glib-HAL_ID1932\src -I.piolibdeps\TMC2130Stepper_ID1493\src -IMarlin\src\HAL\HAL_LPC1768\include -IMarlin\src\HAL\HAL_LPC1768\u8g Marlin\src\HAL\HAL_LPC1768\HAL_timers.cpp
arm-none-eabi-g++ -o .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HardwareSerial.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Os -ffunction-sections -fdata-sections -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m3 -mthumb -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing -fsingle-precision-constant --specs=nano.specs --specs=nosys.specs -MMD -MP -fmax-errors=5 -g -ggdb -DF_CPU=100000000L -DPLATFORMIO=30601 -DU8G_HAL_LINKS -DTARGET_LPC1768 -IMarlin -Iframeworks\CMSIS\LPC1768\include -Iframeworks\CMSIS\LPC1768\lib -Iframeworks\CMSIS -I.piolibdeps\U8glib-HAL_ID1932\src -I.piolibdeps\TMC2130Stepper_ID1493\src -IMarlin\src\HAL\HAL_LPC1768\include -IMarlin\src\HAL\HAL_LPC1768\u8g Marlin\src\HAL\HAL_LPC1768\HardwareSerial.cpp
Marlin\src\HAL\HAL_LPC1768\HAL.cpp:23:10: fatal error: ../../inc/MarlinConfig.h: No such file or directory
#include "../../inc/MarlinConfig.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~
caompilation terminated.rm-none-eabi-g++ -o .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\LPC1768_PWM.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Os -ffunction-sections -fdata-sections -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m3 -mthumb -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing -fsingle-precision-constant --specs=nano.specs --specs=nosys.specs -MMD -MP -fmax-errors=5 -g -ggdb -DF_CPU=100000000L -DPLATFORMIO=30601 -DU8G_HAL_LINKS -DTARGET_LPC1768 -IMarlin -Iframeworks\CMSIS\LPC1768\include -Iframeworks\CMSIS\LPC1768\lib -Iframeworks\CMSIS -I.piolibdeps\U8glib-HAL_ID1932\src -I.piolibdeps\TMC2130Stepper_ID1493\src -IMarlin\src\HAL\HAL_LPC1768\include -IMarlin\src\HAL\HAL_LPC1768\u8g Marlin\src\HAL\HAL_LPC1768\LPC1768_PWM.cpp

arm-none-eabi-g++ -o .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\LPC1768_Servo.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Os -ffunction-sections -fdata-sections -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m3 -mthumb -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing -fsingle-precision-constant --specs=nano.specs --specs=nosys.specs -MMD -MP -fmax-errors=5 -g -ggdb -DF_CPU=100000000L -DPLATFORMIO=30601 -DU8G_HAL_LINKS -DTARGET_LPC1768 -IMarlin -Iframeworks\CMSIS\LPC1768\include -Iframeworks\CMSIS\LPC1768\lib -Iframeworks\CMSIS -I.piolibdeps\U8glib-HAL_ID1932\src -I.piolibdeps\TMC2130Stepper_ID1493\src -IMarlin\src\HAL\HAL_LPC1768\include -IMarlin\src\HAL\HAL_LPC1768\u8g Marlin\src\HAL\HAL_LPC1768\LPC1768_Servo.cpp
Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp:51:10: fatal error: ../../inc/MarlinConfig.h: No such file or directory
#include "../../inc/MarlinConfig.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
arm-none-eabi-g++ -o .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\SoftwareSPI.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Os -ffunction-sections -fdata-sections -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m3 -mthumb -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing -fsingle-precision-constant --specs=nano.specs --specs=nosys.specs -MMD -MP -fmax-errors=5 -g -ggdb -DF_CPU=100000000L -DPLATFORMIO=30601 -DU8G_HAL_LINKS -DTARGET_LPC1768 -IMarlin -Iframeworks\CMSIS\LPC1768\include -Iframeworks\CMSIS\LPC1768\lib -Iframeworks\CMSIS -I.piolibdeps\U8glib-HAL_ID1932\src -I.piolibdeps\TMC2130Stepper_ID1493\src -IMarlin\src\HAL\HAL_LPC1768\include -IMarlin\src\HAL\HAL_LPC1768\u8g Marlin\src\HAL\HAL_LPC1768\SoftwareSPI.cpp
Marlin\src\HAL\HAL_LPC1768\HAL_timers.cpp:31:10: fatal error: ../../inc/MarlinConfig.h: No such file or directory
#include "../../inc/MarlinConfig.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[... cut ...]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment STM32F4                     [SKIP]
Environment teensy20                    [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
========================================================================================= [ERROR] Took 2.80 seconds =========================================================================================

thinkyhead, 2018-10-08:
Unable to reproduce. The includes are correct. Did you delete `MarlinConfig.h`?

timeshell, 2018-12-24:
I am also experiencing something similar as this with a clean download of 2.0.x.  This time it is starting with the file:
Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py

This file isn't in the HAL_LPC1768 folder in Marlin's github right now.

python: can't open file 'Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py': [Errno 2] No such file or directory
OSError: 'python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py -fmax-errors=5' exited 2:
File "C:\Users\gmnut\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\gmnut\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\gmnut\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\gmnut\.platformio\platforms\nxplpc\builder\main.py", line 88:
target_elf = env.BuildProgram()
File "C:\Users\gmnut\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\gmnut\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 104:
env.ProcessFlags(env.get("BUILD_FLAGS"))
File "C:\Users\gmnut\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\gmnut\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 186:
env.Append(**env.ParseFlagsExtended(flags))
File "C:\Users\gmnut\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\gmnut\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 149:
for key, value in env.ParseFlags(str(raw)).items():
File "C:\Users\gmnut\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 800:
do_parse(arg)
File "C:\Users\gmnut\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 676:
arg = self.backtick(arg[1:])
File "C:\Users\gmnut\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 594:
raise OSError("'%s' exited %d" % (command, status))

boelle, 2019-02-21:
@wsidebottom problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead

prusanoff, 2020-03-11:
Problem Not Solved Im Getting this:

`Compiling .pio/build/LPC1768/libb4d/U8glib-HAL_ID1932/U8glib.cpp.o
In file included from Marlin/src/sd/../inc/MarlinConfig.h:44,
                 from Marlin/src/sd/cardreader.cpp:23:
Marlin/src/sd/cardreader.cpp: In function 'void announceOpen(uint8_t, const char*)':
Marlin/src/sd/../inc/../core/utility.h:75:41: error: passing 'const restorer<signed char>' as 'this' argument discards qualifiers [-fpermissive]
   75 | #define RESTORE(N) restorer_##N.restore()
      |                                         ^
Marlin/src/sd/../inc/../core/serial.h:53:33: note: in expansion of macro 'RESTORE'
   53 |   #define _PORT_RESTORE(n)      RESTORE(n)
      |                                 ^~~~~~~
Marlin/src/sd/../inc/../core/serial.h:67:33: note: in expansion of macro '_PORT_RESTORE'
   67 | #define PORT_RESTORE()          _PORT_RESTORE(1)
      |                                 ^~~~~~~~~~~~~
Marlin/src/sd/cardreader.cpp:459:5: note: in expansion of macro 'PORT_RESTORE'
  459 |     PORT_RESTORE();
      |     ^~~~~~~~~~~~
Marlin/src/sd/../inc/../core/utility.h:71:15: note:   in call to 'void restorer<T>::restore() [with T = signed char]'
   71 |   inline void restore() { ref_ = val_; }
      |               ^~~~~~~
Marlin/src/sd/cardreader.cpp: In static member function 'static void CardReader::openFileRead(char*, uint8_t)':
Marlin/src/sd/../inc/../core/utility.h:75:41: error: passing 'const restorer<signed char>' as 'this' argument discards qualifiers [-fpermissive]
   75 | #define RESTORE(N) restorer_##N.restore()
      |                                         ^
Marlin/src/sd/../inc/../core/serial.h:53:33: note: in expansion of macro 'RESTORE'
   53 |   #define _PORT_RESTORE(n)      RESTORE(n)
      |                                 ^~~~~~~
Marlin/src/sd/../inc/../core/serial.h:67:33: note: in expansion of macro '_PORT_RESTORE'
   67 | #define PORT_RESTORE()          _PORT_RESTORE(1)
      |                                 ^~~~~~~~~~~~~
Marlin/src/sd/cardreader.cpp:520:5: note: in expansion of macro 'PORT_RESTORE'
  520 |     PORT_RESTORE();
      |     ^~~~~~~~~~~~
Marlin/src/sd/../inc/../core/utility.h:71:15: note:   in call to 'void restorer<T>::restore() [with T = signed char]'
   71 |   inline void restore() { ref_ = val_; }
      |               ^~~~~~~
Compiling .pio/build/LPC1768/libb4d/U8glib-HAL_ID1932/U8glibPrint.cpp.o
*** [.pio/build/LPC1768/src/src/sd/cardreader.cpp.o] Error 1`

thinkyhead, 2020-03-11:
That compile error has nothing to do with this one. And `PORT_RESTORE` is now patched.