MarlinFirmware/Marlin #11890: ubl anet A8

manotroll, 2018-09-22:
is it possible to use ubl in anet a8?
can the data be stored in sd card?

what would be the way to do automatically with 9 points being 3 on each side of the table and 3 on the middle part?

thinkyhead, 2018-09-28:
> is it possible to use ubl in anet a8?

Yes.

> can the data be stored in sd card?

No. It is stored in EEPROM.

> what would be the way to do automatically with 9 points being 3 on each side of the table and 3 on the middle part?

UBL probes **a regular grid** according to the configured settings:
```cpp
#define UBL_MESH_INSET 0          // Mesh inset margin on print area
#define GRID_MAX_POINTS_X 3       // Don't use more than 15 points per axis, implementation limited.
#define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
```

closed 2018-09-28 by thinkyhead

pastapojken, 2018-12-09:
Unless i'm mistaken(read: enabling it by just uncommenting the ubl define and doing nothing else) the A8 doesn't have enough memory to store Marlin with UBL.