MarlinFirmware/Marlin #11893: TMC 2130 Dual Endstop not Triggering Second X/Y Stepper

Reaction-Time, 2018-09-22:
### Description

Dual endstop functionality does not work when using TMC 2130's in the sensorless homing mode.

Currently attempting to get an MPCNC working with automatic squaring, using sensorless homing instead of endstops. _dual X/Y motors are enabled, so is dual endstops for X/Y_. Dual X/Y motors are working correctly, however, when trying to home/square only one TMC will trigger and will stop only one motor. 
 
### Steps to Reproduce

1. Ramps 1.4 with 4 TMC'S 
2. Set up dual X/Y motors 
3. Set up TMC sensorless home settings
4. Enable dual endstop

**Expected behavior:** For both TMC's to trigger when homing 

**Actual behavior:** One TMC (Either X1 or Y1 depending on axes being homed) will trigger, leaving the X2/Y2 motor spinning and hitting the physical machine min.
#### Additional Information

[MPCNC Marlin 1.9.zip](https://github.com/MarlinFirmware/Marlin/files/2407676/MPCNC.Marlin.1.9.zip)

teemuatlut, 2018-09-22:
I don't think this has been coded yet at this point.
I've been looking more into the whole sensorless homing code and it'll get some updates when I get all the other things done first. I'll see if I can address this issue along the way as well.

Reaction-Time, 2018-09-23:
Perfect, thank you.

mattfredwill, 2019-02-02:
Should be fixed with #13061

boelle, 2019-02-21:
@Reaction-Time problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

mattfredwill, 2019-03-07:
Definitely closable imo, I've been doing the exact use case referenced by the issue (MPCNC with TMC2130s) for the last month with no problems.

closed 2019-05-16 by thinkyhead