MarlinFirmware/Marlin #11894: [2.0.x] Z doesn't move down after home

GMagician, 2018-09-22:
I'm facing a very weird issue.

Before home:
    Z moves both directions (ok never goes below 0 but after a rise I can lower)
After home:
    move rise but doesn't go lowers

Displayed quote increase and decrease correctly, movement is properly loaded into planner (`Planner::_buffer_steps` returns true and `recalculate()` is called) and M119 return all endstop open.

Super weird thing: I loaded new firmware to get this issue but going back with PR it persist.

I'm proceeding to try to understand why planner skip it

AnHardt, 2018-09-22:
Try `ENDSTOP_NOISE_FILTER`.

GMagician, 2018-09-22:
I think I discovered the issue...I left Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN enabled but it shoudn't do anything since I don't have any probe

GMagician, 2018-09-22:
@AnHardt the issue is not during home. Home is done ok, it is after it I can't move Z anymore down

AnHardt, 2018-09-22:
`ENDSTOPS_ALWAYS_ON_DEFAULT` ?

GMagician, 2018-09-22:
No, and using interrupt

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2407948/Marlin.zip)

GMagician, 2018-09-22:
@AnHardt I can move z up 50mm, then m119 (all open), and then down 10mm.
Everything is ok but it doesn't move down..never..
Now I'm cheching inside stepper isr to see why it stop...

GMagician, 2018-09-22:
added a debug test inside stepper isr just after `endstops.update()` call.
my variable is or'ed with `abort_current_block` and reset on M119 call. No abort so endstops are not cause. Also `set_directions` is called this confirm move is queued

GMagician, 2018-09-22:
I'm wondering, even is not my situation since I have DUAL_Z if code below
```cpp
    #elif ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
      UPDATE_ENDSTOP_BIT(Z, MIN);
    #elif Z_HOME_DIR < 0
      UPDATE_ENDSTOP_BIT(Z, MIN);
    #endif
```

should not be:
```cpp
    #elif BED_HAS_PROBE && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
      UPDATE_ENDSTOP_BIT(Z, MIN);
    #elif Z_HOME_DIR < 0
      UPDATE_ENDSTOP_BIT(Z, MIN);
    #endif
```

GMagician, 2018-09-23:
@ejtagle another question:
what is purpose of:
```cpp
  #if DISABLED(ENDSTOP_NOISE_FILTER)
    if (!abort_enabled2()) return;
  #endif
```
inside EndStop.update?

I suspect that if endstops are "globally enabled" you MUST update endstops.

sequence example:
Endstops.enable_globally(true);
Endstops.enable(false)
endstops are always enabled but you don't test them

I'll make a PR to solve my problem and I'll do what you will suggest me, so don't post any PR to fix them

AnHardt, 2018-09-23:
Add DELTA style home adjustments in **every** case. Be sure that adjustment is always bigger than home bump mm for that axis.

GMagician, 2018-09-23:
@AnHardt I think I don't understand (it's not the first time I misunderstand you, please be patient). adjustment is to properly align Z dual stepper, value is mechanical dependent and will move only one stepper and may not be bigger than bump.
Don't know what is "DELTA style" since mine is cartesian and I never worked with delta printers

AnHardt, 2018-09-23:
Deltas do have home adjustments , not only for 1 axis of a pair, but 3 adjustments for 3 theyr 3 axes. If those adjustments are big enough, after homing all hardware endstops are free. This is a requirement for proper working software endstops anyway.

GMagician, 2018-09-23:
On cartesian I think this is not possible. Home is often done approaching bed, endstop are adjusted to be 'hit at pos 0', can't be lower that bed. Delta homes far from bed and can move far from endstops.
My idea is to remove `hit_on_purpose` call from the position it is now and move it after dual alignment and clear also live_state (when interrupt are enabled and endstop are not always tested).
My problem is with G38. I'm not sure what's its purpose. It seems that it bypass endstops when ENDSTOP_NOISE_FILTER is disabled and Endstop on interrupts are enabled.

thinkyhead, 2018-09-24:
> after homing all hardware endstops are free

This must also certainly be the case after we lower the nozzle below the "safe height," which is an optional part of `G28` homing. So homing-validation on Delta will have to be ensured to happen only at the points where endstops are known to be triggered.

closed 2018-09-24 by GMagician