MarlinFirmware/Marlin #11895: TMC2130 SPI communication problem with some CS Pins?!

NetrunnerAT, 2018-09-22:
### Introduction

Sorry for my bad english ... i hope it is enough to understand my issue. 

### Description

I use the Trigorilla Board and change all stepper driver to TMC2130. My Problem is the SPI Communication with some Pins configurated as Chip Select. Possible my problem is a Atmel Mega 2560 issue and not releated to Marlin FW or SPI Lib!! Like some strange register port config issue?

### Steps to Reproduce -> Marlin bugfix 1.1.9 -> TRIGORRILLA_14.h

#define X_CS_PIN           4
#define Y_CS_PIN           5
#define Z_CS_PIN           6
#define Z2_CS_PIN          12
#define E0_CS_PIN          21

Port 21 work well. All TMC2130 report correct with the M122 command. 

#define E0_CS_PIN          11

X = 0xFF:FF:FF:FF
Y = 0xFF:FF:FF:FF
Z = 0xFF:FF:FF:FF
Z2 = 0xFF:FF:FF:FF
E0 = 0x00:00:00:00

#define E0_CS_PIN          19

X = 0xF8:05:00:00
Y = 0xF8:05:00:00
Z = 0xF8:05:00:00
Z2 = 0xF0:05:00:00
E0 = 0xFF:FF:FF:FF

Board definition is clean to prevent double used Pins!

[Marlin_1.1.9_TMC2130_CS-Pin_issue.zip](https://github.com/MarlinFirmware/Marlin/files/2407986/Marlin_1.1.9_TMC2130_CS-Pin_issue.zip)

Palatis, 2018-09-23:
pin19 is ZMAX on trigorilla 14 (ramps 1.4)

NetrunnerAT, 2018-09-24:
I know that. This isnt the point. ZMAX isnt in use in my Setup and Free to use for chip select. Also Pin 11!

Palatis, 2018-09-26:
well if you didn't properly configure your marlin, ZMAX may be initialized for something else (ie.... Z max limit switch...).
for example pull-ups and interrupts may be initialized even if you're not connecting it to anything.

gloomyandy, 2018-09-26:
Might be worth enabling PINS_DEBUGGING and then running M43 this should show if any of the pins are being used for more than one function. http://marlinfw.org/docs/gcode/M043.html