MarlinFirmware/Marlin #11896: LPC1768 error   set motherboard to an LPC1768 motherboard

awesomebender, 2018-09-22:
So I am having a problem  compiling marlin 2.0 using platform.io in atom.io I changed the motherboard to the mks base board and changed the environment to LPC1768 and hit compile and then it had an LPC1768 error I don't know much about programing please help me

forkoz, 2018-09-23:
It would help to copy and paste the error here.

awesomebender, 2018-09-23:
I'm having problems copying the error from atom.io

thinkyhead, 2018-10-26:
Screenshots also work.

gauix4, 2018-10-27:
[image]

gauix4, 2018-10-27:
https://www.youtube.com/watch?v=gMPdsBumWCQ

boelle, 2019-02-21:
@awesomebender problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

PillatV, 2019-03-15:
I had the same issue and spend around may be 1h before i got that configuration located in config folder not used. Please use configuration files located in Marlin folder.

closed 2019-05-16 by thinkyhead