MarlinFirmware/Marlin #11897: Compiles until I enable parking extruder, and or sensorless homing with TMC 2130. Works fine in Marlin 1.1.8.

hallphilip, 2018-09-22:
//file Configuration.h Release:tags/1.1.9
//section info
#define STRING_CONFIG_H_AUTHOR "Phil Hall_DPE_BW"
//section machine
#define MOTHERBOARD BOARD_MKS_GEN_13
#define CUSTOM_MACHINE_NAME "DPE-BW"
#define INVERT_X_DIR true
#define INVERT_Y_DIR false
#define X_BED_SIZE 273
#define Y_BED_SIZE 250
#define Z_MAX_POS 250
//#define MIN_SOFTWARE_ENDSTOP_X 
//#define MIN_SOFTWARE_ENDSTOP_Z 
//section extruder
#define EXTRUDERS 2
#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
#define HOTEND_OFFSET_Z.0 {0.0, 0.00}
#define PARKING_EXTRUDER 
#define PARKING_EXTRUDER_PARKING_X { -71, 273 }
#define PARKING_EXTRUDER_GRAB_DISTANCE 0
#define HOTEND_OFFSET_X {0.0, 71.00}
#define INVERT_E0_DIR true
#define INVERT_E1_DIR true
//section temperature
#define TEMP_SENSOR_0 5
#define TEMP_SENSOR_1 5
#define TEMP_SENSOR_BED 11
#define DEFAULT_Kp.0 20.71
#define DEFAULT_Ki.0 1.59
#define DEFAULT_Kd.0 67.44
#define PIDTEMPBED 
#define DEFAULT_bedKp.0 278.84
#define DEFAULT_bedKi.0 54.55
#define DEFAULT_bedKd.0 356.56
//section homing
//#define USE_XMIN_PLUG 
#define USE_XMAX_PLUG 
#define Z_MIN_ENDSTOP_INVERTING true
#define X_DRIVER_TYPE TMC2130
#define Y_DRIVER_TYPE TMC2130
#define Z_DRIVER_TYPE DRV8825
#define E0_DRIVER_TYPE DRV8825
#define E1_DRIVER_TYPE DRV8825
#define X_HOME_DIR 1
//section motion
#define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 160, 3200, 1666.68 }
#define S_CURVE_ACCELERATION 
//section calibrate
//#define RIGHT_PROBE_BED_POSITION (X_BED_SIZE - (MIN_PROBE_EDGE))
//#define BACK_PROBE_BED_POSITION (Y_BED_SIZE - (MIN_PROBE_EDGE))
#define MESH_INSET.0 10
//section lcd
#define REVERSE_ENCODER_DIRECTION 
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 

//file Configuration_adv.h Release:tags/1.1.9
//section extruder
#define LIN_ADVANCE 
//section homing
#define ENDSTOPS_ALWAYS_ON_DEFAULT 
#define X_HOME_BUMP_MM 0
#define Y_HOME_BUMP_MM 0
#define QUICK_HOME 
//section extras
#define JUNCTION_DEVIATION 
#define MICROSTEP_MODES { 16, 16, 16, 16, 16, 16 }
//section lcd
#define BABYSTEPPING 
#define DOUBLECLICK_FOR_Z_BABYSTEPPING 
//section tmc_smart
#define X_CURRENT 900
#define Y_CURRENT 900
#define TMC_USE_SW_SPI 
#define TMC_SW_MOSI -1
#define TMC_SW_MISO -1
#define TMC_SW_SCK -1
#define MONITOR_DRIVER_STATUS 
#define X_HOMING_SENSITIVITY 5
#define Y_HOMING_SENSITIVITY 5
//#define Z_HOMING_SENSITIVITY 8
#define TMC_DEBUG

hallphilip, 2018-09-22:
I have exhausted all my knowledge of this, hoping someone can help.

Ludy87, 2018-09-23:
Whats is the error message?

Ludy87, 2018-09-23:
SENSORLESS_HOMING requires X_MAX_ENDSTOP_INVERTING and ENDSTOPPULLUP_XMAX when homing to X_MAX.

Change your Code, very easy Error Message

hallphilip, 2018-09-23:
I have tried to no avail, I will try again but have had no luck thus far.

hallphilip, 2018-09-23:
Ok mabe I didn't do that already, but now I am getting this error. Thanks for taking the time to look at this with me.

 
sketch\Marlin_main.cpp: In function 'void tool_change(uint8_t, float, bool)':

Marlin_main.cpp:12484:63: error: too few arguments to function 'void parking_extruder_tool_change(uint8_t, float, bool)'

             parking_extruder_tool_change(tmp_extruder, no_move);

                                                               ^

sketch\Marlin_main.cpp:12319:15: note: declared here

   inline void parking_extruder_tool_change(const uint8_t tmp_extruder, const float fr_mm_s/*=0.0*/, bool no_move/*=false*/) {

               ^

Ludy87, 2018-09-23:
This is a Bug, try Bugfix-1.1.x or Bugfix-2.0.x

hallphilip, 2018-09-23:
I have tried both. scott had made some changes but it is still acting the same way.

hallphilip, 2018-09-23:
thanks Ludy87 IDE must have defaulted back to 1.1.9, once I selected bug fix everything compiled nicely. thank you very much.

Moyster, 2018-09-26:
if your issue is solved, you can close it ;)

closed 2018-09-28 by thinkyhead