MarlinFirmware/Marlin #11901: [1.1.9] Case light doesnt work on MKS GEN L

meav33, 2018-09-23:
LED case lighting doesnt work on the mks gen L board  on the second extruder output (D7) , unless you comment out this line:
#define MOSFET_D_PIN 7 in the pins_mks_gen_L file.
With the line still active, the case light will always turn itself off after about half a  second.

This was unexpected for me, because in the previous releases it did work without having to comment out this line.

dough29, 2018-11-03:
Why not using any other free pin like 4, 5, 6 ,11 (...) ?

thinkyhead, 2018-11-04:
@meav33 — Can you please ZIP and attach your `Configuration.h` and `Configuration_adv.h` file so that we can figure out what `MOSFET_D_PIN` is being assigned to? Thanks.

thinkyhead, 2018-11-04:
@meav33 — First, please check `bugfix-1.1.x` and/or `bugfix-2.0.x` to see if the issue has already been patched. Then, please ZIP up your `Configuration.h` and `Configuration_adv.h` files and drop them in a reply at https://github.com/MarlinFirmware/Marlin/issues/11901 so that we can figure out what `MOSFET_D_PIN` is being assigned to. Thanks.

rarincarp, 2019-02-08:
Hello. I have a very similar problem I can not enable the fans in the lights tells me to use another pin.
do not offend me but I'm Italian and I understand very little English.
I can not even understand what change I have to do to fix the bug you described
ps
I am writing to you through a translator

[Uploading my.config.rar.zip…]()

please be read I read a lot and I lose so many hours but my language does not help me.

rarincarp, 2019-02-08:
I could not sleep tonight! the firmware tormented me.
Resolved
it was a bug and not my mistake

boelle, 2019-02-21:
@meav33 is problem gone in latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead