MarlinFirmware/Marlin #11907: 1.1.9 - Z axis homing halts

Fucazza01, 2018-09-24:
Hello,
I have installed Marlin 1.1.9 on my CR-10s using Arduino IDE 1.8.7.
Unfortunately I'm get always this error of "Homing failed PRINTER HALTED Please reset", before to print a new item.

When I start a new print, the printer execute the "Home" procedure on all axes. If the Z is relatively higt from the End-Point switch, the printer try to arrive at 0, but fail and showing the error message.

I need to repeat the "Home" procedure many times, when the Z finally arrive at the 0 switch everything work fine and I'm able to print the Item.

Steps to reproduce:
1. Select "Prepare" on the control panel
2. Select "Auto Home" on the control panel
2. The X and Y axes go at Home position and press the End-Point switches
3. The Z axes go little bit up, after go down but if is to much higher fail showing the error "Homing failed PRINTER HALTED Please reset".

rmoravcik, 2018-09-24:
The same problem happened to me once or twice, but on X-axis. X was already homed from the previously finished print. And as soon I started a new print I've got the same error.

Fucazza01, 2018-09-24:
Hi,
I have read already about "ENDSTOP_NOISE_FILTER", but the side effect is the poor precision that the flag introduced.
Another possible solution is to soldering a 100nF capacitor to the end-stop, that filter the noise that generate the problem.

I'll try first with the capacitor just ot understand if the filter of the noise will solve the problem.

What is not clear is why this problem appears now and not with the previous version. The Hardware is the same, so that means that the "software noise filter" was already set in the previous version?

AnHardt, 2018-09-24:
> so that means that the "software noise filter" was already set in the previous version?

Exactly, but with a much lower filtering depth. Now 7 before 2.
A quick test in software is much faster to do than soldering in capacitors and easier to reefert. (at least for me)

Fucazza01, 2018-09-24:
Into the code comment I found "...Enabling this feature means adds an error of +/-0.2mm, so homing will end up at a slightly different position on each G28.".
This means that after calibrated the plate, on each print we will have an error of +/-0.2mm on each axes?
If yes, is not better to differentiate the possibility to define on which axes to set this software filtering?

AnHardt, 2018-09-24:
Yes. It's much more complicated, but yes.
Errors in X and Y don't matter at all (as long not resuming), but Z does.
Besides of that, even capacitors 'delay' the signal. If filtering too much, with huge capacitors and a resistor, you get serious problems, because the signal level is too long in the undefined TTL area. Does look than loie a bouncing switch.

Filtering has just to be strong enough to distinguish noise from a starting signal. If that signal is bouncing or not does not matter. The 'old' two **step** filter did it's job surprisingly well.

thinkyhead, 2018-09-29:
> The 'old' two step filter did it's job surprisingly well.

It sure did. It had to do with the way it sampled, I think. It required two "on" signals in a row, and it checked at varying intervals, on each call to the stepper ISR. There might have been more cumulative overhead associated with endstop testing, but since most of the time endstops are turned off it amounted to only one bool test per stepper ISR when not homing / probing.

The new method probably works just as well with the threshold set to 2, but it checks in a different manner and at a different interval. Still, to get as close as possible to the 1.1.8 behavior, it's best to just set the threshold to 2. With that value the accuracy should be very close to that of 1.1.8.

closed 2018-12-30 by thinkyhead