MarlinFirmware/Marlin #11910: ask for help :) i have Question Logo BOOT SCREEN Marlin

ghostlife300, 2018-09-24:
Logo BOOT SCREEN Marlin. Can I change or delete it?
URL Imgur question : https://imgur.com/a/hkX0FK5
If possible Where is the file?

GMagician, 2018-09-24:
Is not possible without coding but I really think that in respect to community it should stay where it is.
If you want you can add your custom logo

Palatis, 2018-09-26:
it's encouraged that manufactures keeps the official marlin logo, and provide a custom bootscreen.
dig through the example config directory for an example.

you can also customize the other icons (such as the nozzle, heatbed, fan), too.

closed 2018-09-28 by thinkyhead

Roxy-3D, 2018-09-28:
Seriously????    You come here and say "I want to use your firmware, but I don't want anybody to know I didn't write or maintain it."      

Are you some kind of @$$ hole?