MarlinFirmware/Marlin #11911: Offset bug, I believe..

renatocarusos, 2018-09-24:
### My offset with M851 does not make the nozzle down to the bed.

I have a mini delta with capacitive sensor probe. 

### To Reproduce
In 1.1.9 ;
    bugfix-1.1.x ;
    bugfix-2.0.x ;

1. G33 P1 ;
2. M211 S0 ;
3. Go down until the nozzle touches the bed ;
4. M851 Z-[how much did he come down] ;
5. M211 S1 ;
6. [still with nozzle leaning on the bed] G33 P0 ;
7. G29 ;
8. M500 ;

**I expected** her to make the offset.

**Actual behavior,** it does not descend, it prints at the z-probe level.

Tank you all.

GMagician, 2018-09-24:
If you try to move up & down Z after G28, it will?
Look at #11894 if you are in the same situation

renatocarusos, 2018-09-24:
@GMagician 
Yes, normally. Even after this routine, both manual and gcode G0 Z0 move the nozzle to the bed, but when it is to print, it stops at the zprobe level [without offset]. 

My slicer is Simplify3D and the "Update Firmware Configuration" option is not enabled, and also the "Global G-Code Offsets" are naturally zeroed.

Some other clue? Ty.

tcm0116, 2018-09-24:
You might need to run `G33 P1` instead of `G33 P0` for step 6. Can you give that a try and report back?

renatocarusos, 2018-09-24:
> You might need to run `G33 P1` instead of `G33 P0` for step 6. Can you give that a try and report back?

[image]