MarlinFirmware/Marlin #11913: TMC2208 Uart not recognized depends on pin

Blamas, 2018-09-24:
Hello, 
I'm using an Mks gen L with 3 tmc2208 clone.
With the actual RX and TX defaut pin defined in the pins_RAMPS.h my drivers can only be registered on the Z (Y and X at 0x00:00:00:00).
I tried to swap the Z and Y pin def in the pins_RAMPS.h and now only the X is registered.

I double check evething and i cannot find where it's coming from, the 3 cables with the resistor are working fine.
Each driver can be registered (when i swap the cable on each one)
I'm using Marlin 1.1.9 and i tried the bugfix, I tried different version of the tmc2208 library and nothing changed.

Actually when i trying to use the motor they are in 1/8 setps (like when you remove all the jumpers on the board)

`Recv: 		X	Y	Z
Recv: Enabled		true	false	false
Recv: Set current	1000	1200	1000
Recv: RMS current	994	1160	994
Recv: MAX current	1402	1636	1402
Recv: Run current	17/31	20/31	17/31
Recv: Hold current	8/31	10/31	8/31
Recv: CS actual		8/31	0/31	0/31
Recv: PWM scale		10	0	0
Recv: vsense		0=.325	0=.325	0=.325
Recv: stealthChop	true	false	false
Recv: msteps		16	256	256
Recv: tstep		1048575	0	0
Recv: pwm
Recv: threshold		76	76	528
Recv: [mm/s]		130.42	2086.72	60.00
Recv: OT prewarn	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false
Recv: off time		5	0	0
Recv: blank time	24	16	16
Recv: hysteresis
Recv: -end		2	-3	-3
Recv: -start		3	1	1
Recv: Stallguard thrs
Recv: DRVSTATUS	X	Y	Z
Recv: stst		X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:	X = 0xC0:08:00:00
Recv: 	Y = 0x00:00:00:00
Recv: 	Z = 0x00:00:00:00`

NSchnitzler, 2018-09-26:
same here, but with the Ramps Board:
```
  #define X_SERIAL_TX_PIN    59
  #define X_SERIAL_RX_PIN    63
  #define Y_SERIAL_TX_PIN    64
  #define Y_SERIAL_RX_PIN    40
  #define Z_SERIAL_TX_PIN    44
  #define Z_SERIAL_RX_PIN    42
  #define E0_SERIAL_TX_PIN   66
  #define E0_SERIAL_RX_PIN   65
```
X and E0 are working, X & Y not. Also switching wires make only the X & E0 registered Stepper controller working. I have also replaced the Mega Board, but with no success.

gloomyandy, 2018-09-26:
Try running an M43 report (you need to enable PINS_DEBUGGING) to check that you don't have pins assigned for more than one use.

Blamas, 2018-09-26:
@gloomyandy You'r right...

'''Recv: .                          X_SERIAL_TX_PIN             protected //Should be 40
Recv: .                          X_SERIAL_RX_PIN             protected //Should be 63

Recv: PIN:  65   Port: K3 (A11)  Y_SERIAL_TX_PIN             Input  = 1 
Recv: .                          Y_SERIAL_RX_PIN             protected  //Should be 42

Recv: PIN:  64   Port: K2 (A10)  Z_SERIAL_TX_PIN             Input  = 0 
Recv: .                          Z_SERIAL_RX_PIN             protected //Should be 59

Recv: PIN:  40   Port: G1        E_MUX0_PIN                  protected
Recv: PIN:  63   Port: K1 (A 9)  Y_CS_PIN                    protected
Recv: PIN:  42   Port: L7        E_MUX1_PIN                  protected
Recv: PIN:  59   Port: F5 (A 5)  FILWIDTH_PIN                protected

How could I define once for all the right ?
Put in comment where the pin are redefined ?

NSchnitzler, 2018-09-26:
Isn't it possible to prevent these issues adding additional sanity checks?

NSchnitzler, 2018-09-27:
you have to keep in mind that "Note: The receive (RX) pins are limited to only interrupt capable pins..." I have switched 40 and 64 and got the Y Axis working. Z Axis is connected to 5 & 6, because Pin 6 is PCINT8, but it still didn't work. M43 tells me:
```
Recv: PIN:   5   Port: E3        Z_SERIAL_TX_PIN                        Output = 1    
TIMER3A   PWM:  -25536    WGM: 0    COM3A: 0    CS: 2    TCCR3A: 0    TCCR3B: 2    TIMSK3: 2   non-standard PWM mode   compare interrupt enabled
Recv: PIN:   6   Port: H3        Z_SERIAL_RX_PIN                        Input  = 1    
TIMER4A   PWM:     0    WGM: 1    COM4A: 0    CS: 3    TCCR4A: 1    TCCR4B: 3    TIMSK4: 0
```
How can i make the last axis working?

Blamas, 2018-09-27:
Don't you need an analog pin (don't remember for which one tx or rx) ? 
I personally just put in comment everywhere the pins are redefined (the AXIS_CS) and my m43 work just fine (not tested yet).

AnHardt, 2018-09-27:
> Don't you need an analog pin (don't remember for which one tx or rx) ?

??? No!
RX needs an interrupt. Ideally a real one (not just 'pin change'). 

Software serial is likely to be incompatible with endstop interrupts if using PCINTs.

thinkyhead, 2018-09-27:
Pin 5 is tied to Timer 3, so it can’t be used for other functions in conjunction with `BLTOUCH` or servo probes.

thinkyhead, 2018-09-27:
> Isn't it possible to prevent these issues adding additional sanity checks?

For example, checking every pin against every other pin? That would be `pins_count!` (factorial) sanity-checks. And, actually, pins are actually allowed to match, as long as only one of the features using that pin number is enabled. So, we have been very selective about sanity-checking pins.

boelle, 2019-02-21:
@Blamas problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead