MarlinFirmware/Marlin #11915: [2.0.x, RFC] Runout sensor enhancement: Trigger run-out after a set distance in mm; add support for encoder disc sensor

marcio-ao, 2018-09-24:
On our branch, I have made two modifications to the filament runout sensor:

1. I added a RUNOUT_DISTANCE_MM configuration parameter that delays the pause of the print until after so many millimeters of filament have been extruded following the run out detection.
2. I added support for a run out sensor that consists of a [magnetic encoder disc](https://www.pololu.com/product/3498) and a hall effect sensor. This differs from an endstop based runout detector in that filament motion causes the FIL_RUNOUT_PIN to toggle, while a run out causes that pin to remain steady.

There are two reasons for the first change: The poles of the encoder disc are about 7mm apart, so the filament must travel that distance in order to ensure the motion is detected. Second, on our printers, the filament sensor is mounted at the end of a feed tube, so if the user wanted to have all the filament in the tube consumed prior to the pause, they could set the RUNOUT_DISTANCE_MM to be the length of the tube.

Adding this code does have a few impacts:

1 - It adds a callback from the stepper ISR each time a block is complete. This allows a step counter to be incremented counting the number of steps taken since the filament run-out was detected.
2 - Adding RUNOUT_DISTANCE_MM adds four bytes per extruder for a step counter.
3 - Making RUNOUT_DISTANCE_MM run-time configurable adds four bytes.
4 - Supporting the encoder disc adds one byte per extruder for tracking state changes, plus one byte of overhead.

Given that this feature does require some resources, let me know whether there would be an interest in incorporating such improvements. If so, I can prepare a PR with my changes.

piercet3dprint, 2018-10-26:
Please also add provision for either an optical or mechanical filament width sensor. Possibly a runout wheel on an arm with an optical wheel for backup. Would need to have a disable option for flexible and transparent fillements.

marcio-ao, 2018-11-15:
@piercet3dprint: The code as it is written should be modular enough for someone to add different types of sensors, but that responsibility will fall to whomever has one of those sensors to write the code.

Anyhow, going to close out this ticket since the basic framework has been added.

closed 2018-11-15 by marcio-ao