MarlinFirmware/Marlin #11918: UBL offset not based on Z end stop position

Cyberwizzard, 2018-09-24:
### Description

I am using UBL using manual probe points. The mesh produced is calibrated using a tool of my own making - it allows me to quickly build a mesh using a laptop hooked onto the printer.

It produced the following mesh (as reported by Marlin):
    Bed Topography Report:
    
        (  1,179)                      (179,179)
            0       1       2       3       4      
     4 | -0.300  -0.530  -0.700  -0.770  -0.720 
       |
     3 | -0.390  -0.590  -0.730  -0.800  -0.810 
       |
     2 | -0.430  -0.650  -0.770  -0.850  -0.880 
       |
     1 | -0.470  -0.700  -0.850  -0.960  -1.010 
       |
     0 | -0.620  -0.770  -0.920  -1.000 [-1.080]
            0       1       2       3       4      
        (  1,  1)                      (179,  1)

However, this mesh does not seem to be used when using the UBL system: the home point for Z at (180,0) is reported to be at Z=0. When loading the mesh anew it is reported to be at 1.08 (which is correct). Moving to Z=0 however, moved the head up from the current position instead of down.

### Steps to Reproduce

< M851 Z0.00
> ok
< G29 L1
> Done.
> ok
< G28 X0 Y0 Z0
> X:180.00 Y:0.00 Z:1.08 E:0.00 Count X:11520 Y:0 Z:0
< G29 L1
> Done.
> ok
< G29 A
> Unified Bed Leveling System v1.01 active.
> ok
< G29 T
*prints topology map as above*
< M114
> X:180.00 Y:0.00 Z:1.08 E:0.00 Count X:11520 Y:0 Z:0
< G01 Z0
*head moves up instead of down*

**Expected behavior:**

In the example above I expect the printer to lower the toolhead 1.08mm below the Z end stop position to touch the bed at (180,0).

**Actual behavior:**

Printer ignores UBL offset and pretends the current position of Z is 0 (and not 1.08) and raises the head instead of lowering it.

#### Additional Information

* Version used: GIT checkout of the 1.1.x branch
* The configuration is for an inductive probe attached to the X carriage; it seems to be not reliable enough for bed leveling (and consistent offsets) so the mesh produced is made by hand.
* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

Cyberwizzard, 2018-09-24:
[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/2412867/Marlin_config.zip)

Uploaded configuration

Cyberwizzard, 2018-09-24:
And by the time I decide to open a ticket it solves itself: 180 is beyond the mesh boundary of 179, so the printer reverts to 0 when reaching X=179.

The only thing that strikes me as odd: the mesh begins at Y=1 up to Y=179, so why does the UBL seem to work for Y=0 & X = <1, 179>? Should it not also go to the original Z = 0 for any Y <= 1?

thinkyhead, 2018-10-26:
Y < 1, sure. Why expect Y==1 to fail if that's within the bounds?

closed 2018-10-26 by thinkyhead