MarlinFirmware/Marlin #11921: [bugfix-2.0.x] software endstops not compatible with UBL

Nitrock28, 2018-09-25:
### Description

When creating the UBL mesh only using the nozzle (G29 P2), with software endstops enabled and Zmin set to 0.0, if the point to probe is below Z=0, the nozzle will never reach the bed as excpected, however the current Z position seen on the LCD actually goes in the negatives, allowing to register a point that is under the bed. By observing closely, the nozzle seems to go down by 1 step below zero, then back up. The steppers actually make some noise when it happens.

For the first few points, i was not very attentive to this behavior and actually registered the points with a negative value, resulting in a mesh forcing the nozzle to sink in the bed during the test print.

Disabling software endstops for probing solved the problem. 

### Steps to Reproduce

1. Launch a UBL manual probing without disabling the software endstops: G29 I 999; G29 P2 H1 T;
2. try to probe a point with a negative Z value.

**Expected behavior:** As the software endstops are on, the nozzle should not go below 0, and the current position should stay at 0

**Actual behavior:** The nozzle goes one step below zero, then back up, decreasing the current Z position each time.

#### Additional Information

BugFix-2.0.X cloned on the 19th september 2018
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/2414388/Configurations.zip)

Nitrock28, 2018-09-25:
First I want to say that I am very new to the whole Marlin and 3D printers source code, so what follows may very well be wrong.

But after some research to locate the origin of this issue, I found something that I find odd and might be linked to the problem. The function  `prepare_move_to_destination` calls `clamp_to_software_endstops` at the very beginning, while `prepare_uninterpolated_move_to_destination` never calls it, and it is that function which is called during the manual probing (Z moves). 

Again, it may be done intentionnally, but I thought it would be a good idea to mention it.

thinkyhead, 2018-10-26:
Indeed. The `prepare_move_to_destination` function was originally designed for use by G-codes like `G1` which should be clamped, and probably shouldn't have become a commonly-used function for other purposes. So, we might want to move the soft endstop clamping out of ther and someplace else, such as `get_destination_from_command` instead.

boelle, 2019-02-21:
@Nitrock28 problem solved?

Nitrock28, 2019-02-22:
Unfortunately, I don't use Marlin Anymore, I recently changed for Klipper.
I didn't have an opportunity to test the current behavior.

boelle, 2019-02-22:
maybe click the close button below then?

closed 2019-02-22 by Nitrock28