MarlinFirmware/Marlin #11922: [bugfix-2.0.x] Number has dissappeared from graphical babystepping screen

hamster65, 2018-09-25:
I flashed marlin 2.0 today and now it only shows the arrows and the nozzle, but no number for the Z offset. It was fine when I flashed 2.0 about 3 days ago.

I am also missing options to adjust acceleration, speed, junction deviation an k-factor during print.

BETLOG, 2018-10-01:
Confirming; missing digits in "Probe Z Offset".

Also; probably related: https://github.com/MarlinFirmware/Marlin/issues/11757

Send: M500
Recv: echo:Settings Stored (609 bytes; crc 39413)
Recv: Error:EEPROM datasize error.

InsanityAutomation, 2018-10-01:
Pr #11973 should have fixed this last night. Can you confirm?

BETLOG, 2018-10-01:
Confirming fixed. (my bad on not checking commit code against what I was using)
Menu screen - Beautiful. Having the digits here is very useful if a little fugly.
Adjustment screen - It was much better with the graphic indicating what meant up and what meant down. The new screen display is a downgrade.
[image]

InsanityAutomation, 2018-10-01:
In order to get the graphical overlay, you need to enable BABYSTEP_ZPROBE_OFFSET as it is dependent on it.

BETLOG, 2018-10-01:
facepalm