MarlinFirmware/Marlin #11924: Can CASE_LIGHT be PWM ?

Roxy-3D, 2018-09-25:
@AnHardt @Bob-the-Kuhn       I have a Case Light on Pin 5 of an AtMega-2560.

M43 Reports:  
PIN:   5   Port: E3            CASE_LIGHT_PIN                         Output = 1    TIMER3A    PWM:  -25536    WGM: 0    COM3A: 0    CS: 2    TCCR3A: 0    TCCR3B: 2    TIMSK3: 2   non-standard PWM mode   compare interrupt enabled

Can I adjust things so I have the ability to set the level?  Or am I stuck with just On and Off ?

thinkyhead, 2018-09-25:
Case light has a brightness setting.
```cpp
SET_OUTPUT(CASE_LIGHT_PIN);
if (USEABLE_HARDWARE_PWM(CASE_LIGHT_PIN))
  analogWrite(CASE_LIGHT_PIN, n10ct);
else {
  const bool s = case_light_on ? !INVERT_CASE_LIGHT : INVERT_CASE_LIGHT;
  WRITE(CASE_LIGHT_PIN, s ? HIGH : LOW);
}
```
```cpp
#if ENABLED(MENU_ITEM_CASE_LIGHT)

  #include "../feature/caselight.h"

  void case_light_menu() {
    START_MENU();
    //
    // ^ Main
    //
    MENU_BACK(MSG_MAIN);
    MENU_ITEM_EDIT_CALLBACK(int8, MSG_CASE_LIGHT_BRIGHTNESS, &case_light_brightness, 0, 255, update_case_light, true);
    MENU_ITEM_EDIT_CALLBACK(bool, MSG_CASE_LIGHT, (bool*)&case_light_on, update_case_light);
    END_MENU();
  }
#endif // MENU_ITEM_CASE_LIGHT
```

Roxy-3D, 2018-09-25:
Yes...  In the case where the pin get gets setup to use PWM to vary brightness.
But Pin 5 is being claimed to be "non-standard PWM mode".   And it only is 100% on or 100% off.

In other words....   USEABLE_HARDWARE_PWM(CASE_LIGHT_PIN) is returning false.

I was hoping I could tweak one of the settings: TIMER3A PWM: -25536 WGM: 0 COM3A: 0 CS: 2 TCCR3A: 0 TCCR3B: 2 TIMSK3: 2    such that USEABLE_HARDWARE_PWM(CASE_LIGHT_PIN) will return true.

boelle, 2019-03-13:
@Roxy-3D should this one still be keept open?

closed 2019-07-11 by shitcreek