MarlinFirmware/Marlin #11925: i am lossing steps when infilling small walls

KARTHICKROBO, 2018-09-25:
i am losing steps   when  printing small walls during infill  at (1 or 2 mm walls ) at 80 to 100mm/s

there is any way to slow down at infill less then 1 mm or 2 mm walls or less ?

if i slow down total infill speed means it will cause big time .

( so only slow down in small walls )

because rapid movement causing lose of steps 

it will save total print !

thank you in advance

GMagician, 2018-09-25:
Marlin can't know when it is doing walls, infill or other things. By it's side is just a generic move

thinkyhead, 2018-09-25:
Try:
- Reducing XY Jerk values
- Reducing XY Acceleration values
- Tuning XY stepper driver Vref current
- Tightening grub screws on XY pulleys

KARTHICKROBO, 2018-09-25:
by the way it increases print time

there is any way we can do in firmware   ??

Sebastianv650, 2018-09-27:
Do you use TMC drivers?

thinkyhead, 2018-09-28:
I've already posted all the troubleshooting advice I can provide without more information.

Please test the `bugfix-1.1.x` and/or `bugfix-2.0.x` branches to see whether it helps. Reduce the Jerk and Acceleration values, and don't forget `M502`, `M500` after flashing the new firmware.

If the problem is resolved by using newer firmware and lowering speeds, then we can close this issue. If the issue isn't resolved, then we can investigate further.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-05-16 by thinkyhead