MarlinFirmware/Marlin #11928: Marlin Compile Errors. too many decimals...

JoeySo, 2018-09-26:
[image]
While trying to compile marlin 1.1.9 from the arduino pc app, when I did the compile/verify with the config files for my CR-10s. Arduino stops (locks up actually) with 100s of syntax errors, such as a number having too many decimal points or missing terminating characters. This sounds like it thinks my config files are corrupt. I tried using the very most recent ones and the little bit older ones from the marlin 1.1.9.zip but still the same errors. I did find though that if I compile with just the original config files in the Marlin1.1.9.zip file, I do not get the errors. Why is it giving me 100s of syntax errors from the CR10s's config files???

Moyster, 2018-09-26:
I'm pretty sure your configuration.h is a html page which you can see by the `<title></title>` error in configuration.h : 30 .  
Guess you didn't "save as" properly, or missed a step ;)

JoeySo, 2018-09-26:
Thanks so much. Yes, I've been playing around with it the last few hours. Reinstalled Arduino Marlin. Anyways, yes the config files got converted to html syntax even though they were still config files. I dont know how it happened. All I can think of is at first, I had the wrong board selected. Maybe that converted them? I dont know. All is fine now. I quess Ill close this now. 
Thanks

closed 2018-09-26 by JoeySo