MarlinFirmware/Marlin #11930: [FR] LinAdv Pattern generator: Loading meshs and G28 after heating up

ikarisan, 2018-09-26:
Hi

This is a feature/change request for the pattern generator found at http://marlinfw.org/tools/lin_advance/k-factor.html 

a) The generated GCODE first homes all axis and after that it heats the bed. In my opinion it makes much more sense to heat up the bed first, wait a few seconds and after that start homing all axis. Or first just home X and Y and after heating up, home Z.

b) You can use G29 to level the bed before printing the pattern. But there should be an alternative option to load a stored mesh data after homing (i.e. M420 L<SELECTED SLOT> S1).

thinkyhead, 2018-09-28:
This really belongs under https://github.com/MarlinFirmware/MarlinDocumentation/issues

forkoz, 2018-09-29:
I concur... I fixed it manually in the gcode. Also had to squeeze bed leveling in there. I think the problem is that the tool is old.

boelle, 2019-02-21:
@ikarisan problem solved?

boelle, 2019-03-13:
@thinkyhead think we can close this one

ikarisan, 2019-03-13:
@boelle Yes :)

boelle, 2019-03-13:
@ikarisan then click the close button below

closed 2019-03-14 by thinkyhead