MarlinFirmware/Marlin #11934: Parser problem? F600 != G0 F600

AnHardt, 2018-09-27:
Currently we get:
```
G1 F600
F600
echo:Unknown command: "F600"
```
As far as i know we should parse that.
All Marlin versions.

AnHardt, 2018-09-27:
The NIST document [The NIST RS274NGC Interpreter - Version 3](https://ws680.nist.gov/publication/get_pdf.cfm?pub_id=823374) says in chapter 3.7.1 page 40: `F` can be used like `T` and `S`, so can stand allone in a code line. Same for LinuxCNC.

`G1 F600` does not do nothing, but sets the feedrate for all following code, until the next `F` is used.
`G1 F600` is just a workaround for the not interpreted `F600` in Marlin and other **printer**-firmwares.
**grbl** and other **CNC**-firmwares do interpret this correctly.

If Marlin wants to be usable in both worlds it has to interpret `F` correctly.

An other improvement Marlin could have is the **modality** of G0, G1, ... , commands. This would make possible lines, with just coordinates, often seen in CNC programs.

As long only slicers are used, made for 3D-printers, there is no problem. They do know about our limitations. As soon as we try to use CNC code generators for mills or lathes (laser cutters?) we are in trouble. 

More a long term project than a acute bug.

thinkyhead, 2018-09-27:
We are very much not following **The NIST RS274NGC Interpreter - Version 3**. Marlin's G-code is only G-code, and it was never modeled on any particular interpreter. If anything, it was modeled on "5D" firmware (with "sticky" parameters).

Marlin has gained a few CNC features, and will be gaining more as time goes on, but it is not likely to be wholly converted into an RS274NGC machine.

TheSFReader, 2018-09-29:
I'm thinking about writing3 "CNC" friendly updates for Marlin : 
- The "parenthese style" comments
- Sticky feedrates
- sticky movement modes
(Of course all of that with configurable options...)

I've implemented the comments in the queue.cpp file (much like the ';' comments)
and think the best way to implement sticky feedrates would be in the GcodeSuite::get_destination_from_command() function

any take on the better place to write the sticky movement mode ?

TheSFReader, 2018-09-29:
Yes that's indeed what I mean by sticky. But I've seen in the MPCNC/V1engineering.com forum more than once that thyere had been problems with Feedrate not being correctly remembered (cf. https://www.v1engineering.com/forum/topic/estlcam-random-dives/#post-12968) but that may be an other problem)

but, well, if feedrate is alredy "sticky" that's so much to not do ;)

TheSFReader, 2018-09-29:
Oh, nevermind, found out why it happens in the case of MPCNC... : Two feedrates are used (XY, and Z) and iif the gcode doesn't repeat the feedrate it uses one fo the other --> inadequate feedrate. I'll remove that from the TODO ;)

tcm0116, 2018-09-29:
The modal movement modes is going to be a bit challenging to implement. Currently, the `GCodeParser` assumes that the command letter and number are provided in every command, and will reject any command that doesn't include the letter and number. As such, the `GCodeParser` would need to be aware of the modality, and might even be the best place to implement it (as it could remember the current movement command and then pass it on to `GCodeSuite`.

tcm0116, 2018-09-30:
>The trick would be to accept X, Y, Z, and F as valid command-codes, all of which get routed to G[last], where last is the last-used G movement code (0, 1, 2, 3, or 5).

That's precisely what I was thinking. Would you clear `G[last]` when a non-movement command-code is received (i.e. `G10`/`G11`)?

TheSFReader, 2018-09-30:
I've written a first implementation of this "modal move", (but will have to adjust it since I hadn't seen the G5 command), and without the chaining/switch-style). 

I updated the parser to remember the last "motion" gcode when a motion mode command is received, and use it when getting an XYZIJKR cfirst letter.

I also updated the queue for the parentheses comments

I don't clear G[last] for now. Should that be desireable ? (RS274NGC doesn't mention it)

A few more questions : RS274NGC pputs also G38 and G80-89 info tha same motion mode. Should I include them ? 

And last : should I include the two features in 2 PR or only one ? and should I use a single #defin/enable marker or both ?

thinkyhead, 2018-10-01:
> That's precisely what I was thinking. Would you clear G[last] when a non-movement command-code is received (i.e. G10/G11)?

I don't think so. According to linuxcnc.org, motion modes are only set by `G0`, `G1`, and `G81`-`G83`. The current motion mode is only canceled by `G80`. The `G53` (machine coordinates) prefix also doesn't require `G0` or `G1` to follow it if the motion mode has already been set.

thinkyhead, 2018-10-01:
> Should I include the two features in 2 PR or only one ? and should I use a single #defin/enable marker or both ?

Both in one PR is fine, and both should be optional and disabled by default.

TheSFReader, 2018-10-01:
I don't see any call to  G53 in 2.0 bugfix. Did I miss anything ?
Also, if I enable "modal motio n" (aka sticky moves) should we store/restore the motion mode that was set before a G53 ?

TheSFReader, 2018-10-08:
@AnHardt this should have been addressed in the #12013 PR which has been merged. Can you please confirm and perhaps close the issue ?

AnHardt, 2018-10-08:
@TheSFReader 
I'll not close before it at least compiles.
And if the author feels no tests are needed before publishing a patch - i feel different.

AnHardt, 2018-10-08:
Some basic tests succeeded after making it compile.

TheSFReader, 2018-10-08:
Oops sorry, will check again.

comps, 2018-10-08:
While on the topic (as other issues like #256 were closed), would it make sense to also add a "cnc-like" mode for G0, so that it always moves at `DEFAULT_MAX_FEEDRATE`?

thinkyhead, 2018-10-08:
I agree with comments on that issue that it’s problematic and should be both optional and accompanied by a strong warning, at least, because moves at high speed increase ringing.

closed 2018-10-08 by AnHardt