MarlinFirmware/Marlin #11942: [bugfix-2.0.x] Two serial ports for RuRamps V1.3?

robustini, 2018-09-28:
### Description

I'm trying to compile the code with the second serial port activated on the "RuRamps4D V1.3", and tells me that it is not supported....... ?

### Steps to Reproduce

Enable the second serial port in the Configuration.h:

#define SERIAL_PORT_2 2

This is the error:

"SERIAL_PORT_2 is not supported for your MOTHERBOARD. Disable it to continue."

#### Additional Information

With Marlin 1.1.8 i've no error and the second serial port work fine.

svs072477, 2018-09-28:
Hi!
I also encountered this problem and added some code to /src/HAL/HAL_DUE/ HAL.h based on code in HAL.h from HAL_LPC1768.h
With these fixes, everything works:
[HAL.zip](https://github.com/MarlinFirmware/Marlin/files/2428354/HAL.zip)

from string 43
//
// Defines
//
#if !WITHIN(SERIAL_PORT, -1, 3)
  #error "SERIAL_PORT must be from -1 to 3"
#endif
#if SERIAL_PORT == -1
//extern USBSerial SerialUSB;
  #define MYSERIAL0 Serial1
#elif SERIAL_PORT == 0
  #define MYSERIAL0 Serial
#elif SERIAL_PORT == 1
  #define MYSERIAL0 Serial1
#elif SERIAL_PORT == 2
  #define MYSERIAL0 Serial2
#elif SERIAL_PORT == 3
  #define MYSERIAL0 Serial3
#endif
  #define NUM_SERIAL 2
  #if SERIAL_PORT_2 == -1
  //extern USBSerial SerialUSB;
    #define MYSERIAL1 Serial1
  #elif SERIAL_PORT_2 == 0
    #define MYSERIAL1 Serial
  #elif SERIAL_PORT_2 == 1
    #define MYSERIAL1 Serial1
  #elif SERIAL_PORT_2 == 2
    #define MYSERIAL1 Serial2
  #elif SERIAL_PORT_2 == 3
    #define MYSERIAL1 Serial3
  #endif/*
#else
  #define NUM_SERIAL 1
#endif    */
//#define MYSERIAL0 customizedSerial

robustini, 2018-09-28:
Work fine, cool thanks!

thinkyhead, 2018-09-28:
Working on a general solution. See #11947.

closed 2018-09-28 by thinkyhead