MarlinFirmware/Marlin #11944: Errors that I don't even know where to start in terms of fixing - help!

GreyArea1966, 2018-09-28:
Do the following errors represent a bug? I have seen other threads posted that seem to indicate they maybe on version 1.8.4 to 1.8.6, but this is happening on the latest version

Sd2Card.h:100:39: error: pasting "/* SPI Master In Slave Out pin*/" and "_DDR" does not give a valid preprocessing token

   #define SPI_MISO_PIN MISO_PIN       // SPI Master In Slave Out pin

                                       ^

C:\Users\*l\AppData\Local\Temp\arduino_build_452372\sketch\fastio.h:94:35: note: in definition of macro '_SET_INPUT'

 #define _SET_INPUT(IO) do {DIO ## IO ## _DDR &= ~_BV(DIO ## IO ## _PIN); } while (0)

                                   ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\Sd2Card.cpp:316:3: note: in expansion of macro 'SET_INPUT'

   SET_INPUT(SPI_MISO_PIN);

   ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\Sd2Card.cpp:316:13: note: in expansion of macro 'SPI_MISO_PIN'

   SET_INPUT(SPI_MISO_PIN);

             ^

Sd2Card.h:100:39: error: pasting "/* SPI Master In Slave Out pin*/" and "_PIN" does not give a valid preprocessing token

   #define SPI_MISO_PIN MISO_PIN       // SPI Master In Slave Out pin

                                       ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\fastio.h:94:61: note: in definition of macro '_SET_INPUT'

 #define _SET_INPUT(IO) do {DIO ## IO ## _DDR &= ~_BV(DIO ## IO ## _PIN); } while (0)

                                                             ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\Sd2Card.cpp:316:3: note: in expansion of macro 'SET_INPUT'

   SET_INPUT(SPI_MISO_PIN);

   ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\Sd2Card.cpp:316:13: note: in expansion of macro 'SPI_MISO_PIN'

   SET_INPUT(SPI_MISO_PIN);

             ^

Sd2Card.h:99:39: error: pasting "/* SPI Master Out Slave In pin*/" and "_DDR" does not give a valid preprocessing token

   #define SPI_MOSI_PIN MOSI_PIN       // SPI Master Out Slave In pin

                                       ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\fastio.h:95:36: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO) do {DIO ## IO ## _DDR |= _BV(DIO ## IO ## _PIN); } while (0)

                                    ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\Sd2Card.cpp:317:3: note: in expansion of macro 'SET_OUTPUT'

   SET_OUTPUT(SPI_MOSI_PIN);

   ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\Sd2Card.cpp:317:14: note: in expansion of macro 'SPI_MOSI_PIN'

   SET_OUTPUT(SPI_MOSI_PIN);

              ^

Sd2Card.h:99:39: error: pasting "/* SPI Master Out Slave In pin*/" and "_PIN" does not give a valid preprocessing token

   #define SPI_MOSI_PIN MOSI_PIN       // SPI Master Out Slave In pin

                                       ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\fastio.h:95:61: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO) do {DIO ## IO ## _DDR |= _BV(DIO ## IO ## _PIN); } while (0)

                                                             ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\Sd2Card.cpp:317:3: note: in expansion of macro 'SET_OUTPUT'

   SET_OUTPUT(SPI_MOSI_PIN);

   ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\Sd2Card.cpp:317:14: note: in expansion of macro 'SPI_MOSI_PIN'

   SET_OUTPUT(SPI_MOSI_PIN);

              ^

Sd2Card.h:101:39: error: pasting "/* SPI Clock pin*/" and "_DDR" does not give a valid preprocessing token

   #define SPI_SCK_PIN SCK_PIN         // SPI Clock pin

                                       ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\fastio.h:95:36: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO) do {DIO ## IO ## _DDR |= _BV(DIO ## IO ## _PIN); } while (0)

                                    ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\Sd2Card.cpp:318:3: note: in expansion of macro 'SET_OUTPUT'

   SET_OUTPUT(SPI_SCK_PIN);

   ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\Sd2Card.cpp:318:14: note: in expansion of macro 'SPI_SCK_PIN'

   SET_OUTPUT(SPI_SCK_PIN);

              ^

Sd2Card.h:101:39: error: pasting "/* SPI Clock pin*/" and "_PIN" does not give a valid preprocessing token

   #define SPI_SCK_PIN SCK_PIN         // SPI Clock pin

                                       ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\fastio.h:95:61: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO) do {DIO ## IO ## _DDR |= _BV(DIO ## IO ## _PIN); } while (0)

                                                             ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\Sd2Card.cpp:318:3: note: in expansion of macro 'SET_OUTPUT'

   SET_OUTPUT(SPI_SCK_PIN);

   ^

C:\Users\*\AppData\Local\Temp\arduino_build_452372\sketch\Sd2Card.cpp:318:14: note: in expansion of macro 'SPI_SCK_PIN'

   SET_OUTPUT(SPI_SCK_PIN);

              ^

Using library LiquidCrystal at version 1.0.7 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1
pasting "/* SPI Master In Slave Out pin*/" and "_DDR" does not give a valid preprocessing token

TheSFReader, 2018-09-28:
As far as I understand, could be caused by the pre-compiler which doesnt't like comments at the end of #define lines.
Can you try adding an carriage return juste before the comments in the lines ?

GreyArea1966, 2018-09-28:
Thank you I will give that a go

GreyArea1966, 2018-09-28:
You, sir, are a genius. Problem solved.

TheSFReader, 2018-09-28:
Glad to help :)

A Genius ? Not. Just remembered reading some issues and their fixes :) 

https://github.com/MarlinFirmware/Marlin/issues/10862  (May I suggest you close this one now that it's solved ?)

GreyArea1966, 2018-09-28:
I will...though unfortuinately my SD card slot has stopped working...dunno if it's related?

GreyArea1966, 2018-09-28:
Fixed it...just had to initialise it...will close this one

closed 2018-09-28 by GreyArea1966

thinkyhead, 2018-09-28:
Arduino 1.8.7 has fixed this bug.

GreyArea1966, 2018-09-28:
Seeing as I just installed 1.8.9...I'm gonna go with...nope.

Phr3d13, 2018-09-28:
Are you using a dev/nightly/etc version? Latest release version is 1.8.7

GreyArea1966, 2018-09-28:
I'm using this one...novice with github, but I thought it was the latest full release?

https://github.com/MarlinFirmware/Marlin

[image]

Phr3d13, 2018-09-28:
What I'm getting at is, as of the time of writing this, there is no 1.8.9 version of either arduino ide or Marlin firmware. Arduino ide release version is at 1.8.7, and latest release version of Marlin is 1.1.9

thinkyhead, 2018-09-28:
> I thought it was the latest full release?

Um, that's MARLIN there.
I specifically said "_**Arduino 1.8.7 has fixed this bug.**_"

GreyArea1966, 2018-09-29:
Sorry...was late, didn't read properly...will upgrade IDE at earliest opportunity...