MarlinFirmware/Marlin #11945: [1.1.9] Negative Y_PROBE_OFFSET_FROM_EXTRUDER, assertion error on compilation

tsmaa, 2018-09-28:
### Description

When setting up a 3dtouch in front of the nozzle with negative Z_probe offset, assertion happened on compilation
### Steps to Reproduce

1. [First Step]
   On 300x300x400 Creality CR-10 Bed dimension with 3dtouch or bltouch,
 (file configuration.h)

   #define X_PROBE_OFFSET_FROM_EXTRUDER 0  // X offset: -left  +right  [of the nozzle]
   #define Y_PROBE_OFFSET_FROM_EXTRUDER -43  // Y offset: -front +behind [the nozzle]

2. [Second Step]
   Assertion error happened from sanitycheck.h
static_assert(BACK_PROBE_BED_POSITION <= MAX_PROBE_Y, "BACK_PROBE_BED_POSITION is outside the probe region.");

**Expected behavior:** [What you expect to happen]
 no assertion/error on compilation

**Actual behavior:** [What actually happens]
Assertion error

#### Additional Information
This is not a bug fix !!!
The workaround to this issue is: 

Step 1. Place double slash to comment line 949 (sanitycheck.h)
// static_assert(BACK_PROBE_BED_POSITION <= MAX_PROBE_Y, "BACK_PROBE_BED_POSITION is outside the probe region.");
Step 2. Place a double slash to comment line 4993 (marlin_main.cpp)
//          return;

WARING !!! Modification on production 3dprinter is at your own risk. double-check offsets and dimensions before compilation !

thinkyhead, 2018-09-29:
You totally forgot to attach your configurations.

tsmaa, 2018-09-29:
Configuration files attached for checking.
Have a nice day

[TSMAA Marlin Bugfix-1-1-9.zip](https://github.com/MarlinFirmware/Marlin/files/2430432/TSMAA.Marlin.Bugfix-1-1-9.zip)

tsmaa, 2019-01-23:
Close this thread, please.
Issue solved on Marlin bugfix 1.1.9 source downloaded on January 7, 2019
How ? i don't know....

Thank you
Have a nice day

closed 2019-01-23 by tsmaa