MarlinFirmware/Marlin #11955: Alcune domande probabilmente banali

mr-miky, 2018-09-29:
I'm trying to run the display (chipset st7920) directly with a spi hardware (STM32F405) without necessarily doing bitbangig of the pins.

From what I understand I have to define U8G_HAL_LINKS.
I do not understand whether I should include the U8glib library, or U8glib-HAL / U8glib-HAL-dev. With the other two the compilation fails with errors that at the moment I do not remember well.

I tried with the first one, the compilation is successful but the display is completely dead.

I took a look at the HAL of the LPC1768 and there is u8g_com_HAL_LPC1768_st7920_hw_spi.cpp, I suppose I have to implement something equivalent for STM32F4.

Can someone give me some advice on what I have to actually implement in order not to become crazy?

Thank you

 Marlin 2.0 .x  bugfix update to head ...

thinkyhead, 2018-09-29:
> I took a look at the HAL of the LPC1768 and there is `u8g_com_HAL_LPC1768_st7920_hw_spi.cpp`, I suppose I have to implement something equivalent for STM32F4.

That's basically it. Start by duplicating the LPC1768 stuff over to the STM32F4 HAL and build on it. You may be able to ask @Bob-the-Kuhn for some additional guidance.

thinkyhead, 2018-10-03:
Arduino STM32 is now an option.

xC0000005, 2018-10-04:
STM32 generic doesn't init systick before some constructors without a fix. If you insist on using that core (and my M200 V2 support is currently on it, since i haven't managed to make it work with the official core yet), clone STM32GENERIC from my repo, which at least has that fixed + a few others. The official core is coming along. You can build using the Arduino IDE and edit in PIO/whatever you like.

mr-miky, 2018-10-04:
Thank you. I do not insist on using STM32Generic but at the moment when I use official STM32 I get some code that does not work on the board I designed.
I added a bit of #ifdef to the marlin code since the serial changes name and something else. The build of the source ends correctly but then nothing works. I do not like the management of pin names in the official version. I have to edit variant.h and variant.cpp, so that PA0 is associated with D0 and then D0 is translated into PA_0 through enumerations and arrays. It seems to me only a waste of resources.

jmz52, 2018-12-03:
You might want to look at https://github.com/MarlinFirmware/U8glib-HAL/blob/dev/src/clib/u8g_com_stm32duino_hw_spi.cpp
It is an implementation of communication function for MKS Mini 12864 using HW SPI interface (STM32F1/F4 HALs)
It works without U8G_HAL_LINKS.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-13 by mr-miky