MarlinFirmware/Marlin #11956: Unable to Compile for MKS SBase

Bobblejot, 2018-09-29:
### Description

Unable to compile using Platform IO for the MKS SBASE after updating to the latest version of Marlin 2.0 Bugfix today and also to the latest version of Platform IO. Previously everything was compiling fine.

Marlin 2.0 Bugfix 
Platform IO 1.0.2 Core 3.6.1a4
NXP LPC 3.4.1

Issues

1. E2END error ( Have now defined in pins file - #define E2END 0xFFF )
2. Missing Semi Colon ( M104_M109.cpp Line 93 - set_temp = no_wait_for_cooling || parser.seenval('R'); )

3. arm-none-eabi-g++: fatal error: /Users/~/.platformio/packages/toolchain-gccarmnoneeabi/bin/../l
ib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/nano.specs: attempt to rename spec 'link' to already defined spec 'nano_link'

This last issue i cannot fix myself and to be honest don't know if the previous issues were fixed properly either :-)

p3p, 2018-09-29:
Always supply your configs as requested in the issue template, 

1. E2END should not be required by Marlin core anymore it been pushed back behind the HAL, are you trying to run the MKS SBASE with a physical EEPROM chip (spi, i2c)?

2. yup it will be fixed soon

3. you can't (shouldn't be able to) be using NXP LPC 3.4.1 as the platformio.ini file specifies (`platform          = nxplpc@<3.4.0`), the linking issue is a know problem with their newer scripts.

p3p, 2018-09-29:
Can you redownload, you seem to have a corrupt version of Marlin.

thinkyhead, 2018-09-30:
Built for MKS SBASE earlier today. No issues.

boelle, 2019-02-21:
@Bobblejot problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-15 by p3p