MarlinFirmware/Marlin #11962: Temp LEDs

ruggb, 2018-09-29:
I have the temp LEDs configured on my unit. It works fine in normal operation, but if I do an M303 for the bed or the hotend, it is not functional. Is that an oversight or design intent. If I expecting it to tell me when the device is hot, I would think it should function all the time.

thinkyhead, 2018-09-30:
Correct, they are not used within `M303`. As `M303` is meant to be used very seldom and its code is meant to be tight and singularly-focused for the purpose of tuning a new hot-end, interspersing temp LED code into the `M303` algorithm is a very low priority.

GMagician, 2018-10-10:
@ruggb may you test above PR?

thinkyhead, 2018-10-10:
I would, but I don't have a machine with LED lighting.

ruggb, 2018-10-10:
OK I think I got the right one - 6134be - correct?

I'm working on it

GMagician, 2018-10-10:
6134be is just a part of it.  I think you have to clone mine branch but I don't know how...

ruggb, 2018-10-10:
OK, what should I expect? Am I using the correct build? Because it is the same as b4.
When I run M303 e0 s60, the hotend turns on but no LEDs.
If I reset the system (it does not respond to turning off the hotend when running M303) the red LEDs come on as it is cooling down and turn green at ~53.5°C. If it is green and I run m303 again it stays green. I guess I would expect it to set it red b4 starting the routine, then once it finishes it looks like the normal code would take care of turning it green at the proper temp.
If I just turn on the hotend or the bed, red lights up as soon as it is turned on even though it is under 50°C.

ruggb, 2018-10-10:
to be clear, my config uses an RGB led and I am driving it with 2 pins, 1 for red and 1 for green. Not sure that makes a difference.

thinkyhead, 2018-10-11:
Assign your green pin to the blue pin and the printer event LED for heating will transition from green to yellow to red rather than from blue to violet to red.

boelle, 2019-03-13:
@ruggb still waiting for this one?

closed 2019-03-14 by ruggb