MarlinFirmware/Marlin #11967: Acceleration times not equal using S-Curve

Sebastianv650, 2018-09-30:
Using 2.0 bugfix with AVR board. Notifying @ejtagle.

Since the day of the PR for S-Curve acceleration I noticed slightly reduced print times using it compared to the exact same configuration without S-Curve. In theory, this shouldn't be the case as S-Curve tries to achieve the set acceleration value as the average acceleration. See this graph, better than any text explanation: https://github.com/MarlinFirmware/Marlin/pull/11311#issuecomment-406363946.

I forgot about that until today, where I measured the vibration of my printer with an accelerometer just for curiosity.
Here is a comparison between a stop from 175mm/s to 0 on my Y axis, both at 1100mm/s². Both using the same start and stop point and are recorded multiple times so I can say this is perfectly consistant.

Both charts are scaled the same and aligned to the start point of the deceleration phase. Green boxes are deceleration itself, red is "ringing" after the axis has stopped. Things that can be seen:
- The duration of the acceleration is 0,16s with classis acceleration compared to 0,125s with s-curve (78%)
- The noise (vibration) during classic acceleration is significant, with spikes up to 2280mm/s². S-Curve reaches 2420mm/s² max, but with a lot less vibration.
- Ringing takes 0,12s until it reaches vibration levels from before the acceleration phase, with a starting amplitude of 1300mm/s². With S-curve, this is reduced to 0,09s and 730mm/s² - even with the faster than intended acceleration!

[image]

I guess the question is: Is this faster acceleration compared to linear acceleration a bug, or the result of the approximation which is done for the AVR?

With or without bug, I was very happy to see such a huge difference between the two modes, with s-curve as a clear winner!

GMagician, 2018-09-30:
Starting from a equal speed (sure both reached 175mm/s?) and equal max acc (I hope this used in linear part of S ramp as it is used in T one), I'm expecting a longer ramp, not a shorter.
The only reason for a shorter ramp time is a lower reached speed....or a bug

Sebastianv650, 2018-09-30:
I used a 200mm long move, no doubt it reaches full speed. It's also visible in the full plot that the printer accelerates, cruses and decelerates again. The section above is only showing a fraction of the cruising phase, deceleration and than the axis at full stop.

> equal max acc (I hope this used in linear part of S ramp as it is used in T one)

No, S-curve is not using the acceleration as set by M204 as max. acceleration but as average accel.

ejtagle, 2018-09-30:
I do suspect the faster acceleration could be caused by rounding errors and finite resolution of the stepper ISR... Yes, S-Curves are approximate ... ;)

boelle, 2019-02-21:
@Sebastianv650 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.
Before posting a bug report please test with `bugfix-2.0.x` to check if problem is gone

Sebastianv650, 2019-02-21:
@boelle how do you think was that old topic not a development related question as I was changing and debugging the acceleration movement code in Marlin that time :-D

But it's outdated nevertheless, as the acceleration time diferences are caused by the approximation as stated by ejtagle. Closing.

closed 2019-02-21 by Sebastianv650