MarlinFirmware/Marlin #11968: [Bugfix-2.0.x] planner.cpp (20652)

mr-miky, 2018-09-30:
Buid fails with :
`macro "MAX" passed 4 arguments, but takes just 2`

mr-miky, 2018-09-30:
I have temporarily added to planner.cpp
`#define MAX4(a,b,c,d) MAX(MAX(a,b),MAX(b,c))` 
 but I do not know if it is the best solution.

chrissbarr, 2018-10-01:
I'm seeing this too - compilation fails using the STM32F4 HAL / STM32Generic Arduino core, because the definition for MAX in STM32Generic seems to be overriding the new MIN/MAX functions.

If I revert the bugfix-2.0.x branch to the commit before b30ca652aed417be0ba8857922c1cf2b4df26f2d it compiles without issue.

I also don't know what the solution is here - the new MIN/MAX functions look like a cool simplification - maybe we can get the conflicting stuff in STM32Generic renamed?

thinkyhead, 2018-10-01:
Can you describe your build procedure? I've never done a build for STM32F4.

chrissbarr, 2018-10-01:
No worries, I'll detail what I'm doing - it should be straightforward to replicate. I'm running W10 x64, Arduino IDE 1.8.5. I'm using forked versions of STM32GENERIC and Marlin while I work on my own control board, but the problem can be replicated with the base repos:

- [STM32GENERIC Arduino Core](https://github.com/danieleff/STM32GENERIC) downloaded and installed to Arduino/hardware folder. 
- Marlin configuration.h modified so:
  - #define MOTHERBOARD BOARD_STM32F4
  - #define SERIAL_PORT -1
- Arduino options in Tools menu:
  - "Discovery F407VG" board selected
  - USB set to "Serial [Virtual COM Port]"
  - Serial Communication set to "SerialUSB"

Before b30ca652aed417be0ba8857922c1cf2b4df26f2d , the above compiles without issue. After, there's a conflict with the multiple definitions of the MAX macro, where it seems that the definition in STM32GENERIC (which is limited to two arguments) is being used over Marlin's new version:

`C:\build\sketch\src\module\planner.cpp:1765:105: error: macro "MAX" passed 4 arguments, but takes just 2`
`block->step_event_count = MAX(block->steps[A_AXIS], block->steps[B_AXIS], block->steps[C_AXIS], esteps);`

Let me know if there's any other setup/build info you need.

thinkyhead, 2018-10-01:
Ok, I guess I'm doing everything right then. I just needed to disable warnings to stop the torrent of messages. Simplest thing we can do is just rename our macros.

closed 2018-10-01 by thinkyhead

mr-miky, 2018-10-01:
Perfect, now the build ends without errors.

My configuration:
Eclipse (Sloeber addon), STM32generic, STM32F407 , custom board

chrissbarr, 2018-10-01:
Yeah, working great for me now too. Also appreciate the reduced warnings, cheers!

thinkyhead, 2018-10-01:
And didn't need to rename them, either… I had a PR with that kind of change all ready to go, but got to thinking about the inclusion order and the fact that only Marlin code needs the variadic versions.

The move of `#include "minmax.h"` over to `MarlinConfig.h` points to a set of changes that would be useful to get better organized. All Marlin code that needs configuration options includes `MarlinConfigPre.h`, and this also includes `macros.h` because the config files depend on `ENABLED`, `DISABLED`, and a couple of other macros.

So, I'm thinking that the macros the configurations need ought to be moved to `core/config.h` and that should be included in `MarlinConfigPre.h` instead of the full `macros.h` file to avoid stepping on same-named macros that some library headers define. And then `macros.h` can be included at the end of `MarlinConfig.h` (which Marlin code relies on), replacing `minmax.h`.

Then, if any Marlin `.h` files need to use Marlin-specific macros we'll just ensure they include `MarlinConfig.h` and not just `MarlinConfigPre.h`. Anything in the `HAL` folder that isn't a contributed lib can be counted as "Marlin code" under this scheme.

It's not a huge change, but it should help to avoid the accidental poisoning of —or conflicts with— any library `#include`s that happen to follow `#include "MarlinConfigPre.h"`.

mr-miky, 2018-10-02:
Not working again ....

thinkyhead, 2018-10-03:
@mr-miky — Builds are passing for all platforms on our end.

thinkyhead, 2018-10-03:
Should be working now for STM32F4.

mr-miky, 2018-10-03:
I'll check it better tomorrow evening