MarlinFirmware/Marlin #11969: home only on 1 \ 4 of the bed ..  marlin 1.1.9

marco1703, 2018-09-30:
### Description

all that has to do from the home onwards does it in 10 x 10 instead of in the whole surface of the bed ... I try to attach a video

marlin 1.1.9

bltouch

mks 1.4 gen

bed 220 x 220

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

marco1703, 2018-09-30:
[image]

p3p, 2018-09-30:
Always supply your configs as requested in the issue template.

marco1703, 2018-09-30:
Sorry

Marlin 1.1.9-mks gen 1.4- tmc2100-220x220x240- motor 0,9 deg 1,7 amp

p3p, 2018-09-30:
> Additional Information
> - Include a ZIP file containing your Configuration.h and Configuration_adv.h files.

We need the configuration files used to build Marlin in order to start to help you.

marco1703, 2018-09-30:
Ok one moment

Roxy-3D, 2018-10-02:
> marco1703 commented 2 days ago
> Ok one moment

Two days ago is 'one moment' ?   We'll get back to you in a minute or 'day or two' when we have the configuration files.

closed 2018-10-02 by Roxy-3D