MarlinFirmware/Marlin #11974: Platform .IO compile errors

CAP1Sup, 2018-09-30:
Platform.io gives a weird bug when compiling for Marlin. I have tried compiling in Arduino IDE, and after replacing some of the files with the ones that are attached, the Arduino IDE compiled it fine. However, Platform.io is giving some weird errors in the terminal.

[Configuration_Files.zip](https://github.com/MarlinFirmware/Marlin/files/2431750/Configuration_Files.zip)
[arduino-builder-249.zip](https://github.com/MarlinFirmware/Marlin/files/2431752/arduino-builder-249.zip)

Here is the terminal:
> Executing task in folder Marlin-bugfix-2.0.x: C:\Users\xmanp\.platformio\penv\Scripts\platformio.exe run <

Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing c1921b4
Downloading  [####################################]  100%
Unpacking  [######------------------------------]   17%
Unpacking...
Error: Traceback (most recent call last):
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 107, in cli
    result = (envname, ep.process())
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 205, in process
    result = self._run()
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 300, in _run
    self.verbose)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 323, in _autoinstall_libdeps
    ctx.invoke(cmd_lib_install, libraries=[lib], silent=not verbose)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\platformio\commands\lib.py", line 103, in lib_install
    library, silent=silent, interactive=interactive, force=force)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 352, in install
    force=force)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 718, in install
    name, url, requirements, track=True)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 444, in _install_from_url
    self.unpack(dlpath, tmp_dir)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 224, in unpack
    return fu.unpack(dest_dir, with_progress=False)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 112, in unpack
    self._unpacker.extract_item(item, dest_dir)
  File "c:\users\xmanp\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 38, in extract_item
    self._afo.extract(item, dest_dir)
  File "C:\Python27\Lib\zipfile.py", line 1028, in extract
    return self._extract_member(member, path, pwd)
  File "C:\Python27\Lib\zipfile.py", line 1083, in _extract_member
    file(targetpath, "wb") as target:
IOError: [Errno 2] No such file or directory: 'c:\\Users\\xmanp\\Desktop\\3-D printing\\Anet A8\\Upgrades\\Hardware\\Marlin-bugfix-2.0.x-TEST4\\Marlin-bugfix-2.0.x\\.piolibdeps\\_tmp_installing-w6wmgz-package\\TMC26XStepper-c1921b4856a5ef4b659b661b8cb5dd56f3075e57\\documentation\\html\\_t_m_c26_x_stepper_8cpp_source.html'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

CAP1Sup, 2018-09-30:
The discussion for the Arduino fix is here:
https://github.com/olikraus/U8glib_Arduino/issues/9

Ludy87, 2018-09-30:
This is a problem of PIO

try 

    pio update & pio upgrade

thinkyhead, 2018-10-01:
You may just want to try a fresh new install of PlatformIO.

CAP1Sup, 2018-10-02:
I think that the TMC26xx library isn't installing. I have checked the installed libraries, and the TMC26xx library isn't on that list.

thinkyhead, 2018-10-02:
It's downloading for me. It ends up in the folder `.piolibdeps/c1921b4`.

CAP1Sup, 2018-10-02:
The issue might be there. Could you post a folder tree with names? That way I could check it with what I have.

amigoloko, 2018-11-08:
> The discussion for the Arduino fix is here:
> [olikraus/U8glib_Arduino#9](https://github.com/olikraus/U8glib_Arduino/issues/9)

move entire directory of the project as near as possible to C:/, or any other removable storage device. it helps also if rename folders for shorter versions

thinkyhead, 2018-11-08:
@amigoloko — Just use Arduino 1.9 beta for Windows which can handle the longer command line generated by having so many more files.