MarlinFirmware/Marlin #11975: 255 points to 10,000??

luisluna2, 2018-09-30:
Hello friends, I have interesting in modify Firmware Marlin for some proposes, my one specific question is:

how can INCREASE THE "255 POINTS" TO for example "10,000 POINTS" IN MARLIN FOR EXTRUDER, BED AND FAN?

thinkyhead, 2018-10-01:
What is it that you are trying to do? You should explain at length in your native language and let us run it through Google Translate to get a better understanding of your intentions.

mr-miky, 2018-10-01:
Or it is Italian and meant ten point zero zero zero (in Italian we use the point as a separator for the thousands and the comma for the decimals). It would make sense if intended as a 0-10V control, but this is just my guess.

thinkyhead, 2018-10-01:
Fans connected to a board might be 5V, 12V, or 24V.

mr-miky, 2018-10-01:
It is a typical convention for some proportional analog controls 0V = off, 10V full power 5V = half etc etc.
Maybe he has GCODE with 0-10V but I'm still hypothesizing.

GMagician, 2018-10-01:
It will be better to think about 0-100% and convert it to dac 0-255. This will be voltage independent,
but no host software will use it I suspect.

luisluna2, 2018-10-01:
Hablo Español, disculpen mi mal inglés, lo que intento hacer es modificar en Marlin el S del ventilador de S0-255 a S0-10,000.

Digamos que sólo quiero cambiar el 255 por el valor 10,000 para alcanzar el 100%

Entonces quedaría así: 

S0 = 0V
S10,000=10V
S5000= 5V

thinkyhead, 2018-10-02:
Your G-code won't be compatible with any known slicers, but you can accomplish this by altering the code in `GcodeSuite::M106`.
```diff
-   const uint16_t s = parser.ushortval('S', 255);
+   const uint16_t s = map(parser.ushortval('S', 255), 0, 10000, 0, 255);
```
(You won't be allowed to use the comma, as in `S10,000`. You must use `S10000` instead.)

thinkyhead, 2018-10-02:
Your G-code won't be compatible with any known slicers, but you can accomplish this by altering the code in `GcodeSuite::M106`.
```diff
-   const uint16_t s = parser.ushortval('S', 255);
+   const uint16_t s = map(parser.ushortval('S', 10000), 0, 10000, 0, 255);
```
(You won't be allowed to use the comma, as in `S10,000`. You must use `S10000` instead.)

luisluna2, 2018-10-02:
@thinkyhead Friend, in what file need modify?

luisluna2, 2018-10-03:
All Marlin's are different, in my case no exist the red line like you was put, but Now I can check M106 Command, exist a similar line. I check in 4 versions of Malin.

Thank you for you Help.

thinkyhead, 2018-10-03:
It's in the `GcodeSuite::M106` method in Marlin 2.0.x (`M106_M107.cpp`) or the `gcode_M106` function in Marlin 1.1.x (`Marlin_main.cpp`). The line is identical in both.