MarlinFirmware/Marlin #11985: Contour Crafting

jot4p, 2018-10-02:
Hi, 

I have a rectangular extruder for large printed lines. 

It's possible to use marlin to create a contour crafting system ? 

like this: https://youtu.be/1Pg4YVi_Q-M?t=36s 

What is the principle ? follow the lines direction ? where is this on the marlin code ? 

Best Regards.

Grogyan, 2018-10-02:
That video shows some custom code to indepenantly rotate the head from the print direction.
Both Smoothieware and 32bit RepRap firmware support custom scripts for these sorts of scenarios
[https://www.youtube.com/watch?v=bn4gWYOzHxQ](E3D Tool Changer)

thinkyhead, 2018-10-03:
There's no such thing in Marlin at this time. Someone would have to implement support for an extra independent stepper motor as an additional axis. Once that was added, it would be possible to also add code that keeps the extra stepper somewhat tangent to the current move, or for the slicer to include extra information to do that to save on computation in the firmware. It's a relatively big undertaking, on par with the `HANGPRINTER` support added in 1.1.9.

AnHardt, 2018-10-03:
That could be done with Marlin. I see a pretty normal XYZE printer with 3 servos (rotation, flap, flap). Should be manageable with very custom g-code. No advanced automatisms needed.

thinkyhead, 2018-10-03:
I suppose a servo could work. If the nozzle is square it only needs 90° of rotational freedom.

AnHardt, 2018-10-03:
With the right gearing even 720° is no problem.

jot4p, 2018-10-03:
Yeah... I think that is possible with "special" slicer to include code for "E Axis" and after that configure the marlin for 1 more axis in E with 1 servo. 

But I need that slicer. Article about this:
https://pdfs.semanticscholar.org/3e1a/6c271af8a8bd7ae510ceb62f6ca63b46b16c.pdf

jot4p, 2018-10-03:
> .. 3 servos (rotation, flap, flap). Should be manageable with very custom g-code. .. 

the side flaps are not very important, but the rotation of the E-axis is very interesting to use rectangular extruders with 4x ... 10x more material

jot4p, 2018-10-10:
any idea of slicer for this E axis with contour crafting ??