MarlinFirmware/Marlin #11989: [Bugfix-2.0.x] Error:EEPROM datasize error. Error:Field planner_extruder_advance_K mismatch.

Ludy87, 2018-10-03:
I try in Arduino IDE v1.9 compile commit https://github.com/MarlinFirmware/Marlin/commit/c3229e1b3461b6da8373e7a24a7eeb131912a15b - it's broken with:

    In file included from 
    C:\Users\Ludy\AppData\Local\Temp\arduino_build_799868\sketch\src\feature\tmc_util.h:28:0,
                 from 
    C:\Users\Ludy\AppData\Local\Temp\arduino_build_799868\sketch\src\feature\tmc_util.cpp:27:
    C:\Users\Ludy\Documents\Arduino\libraries\TMCStepper\src/TMCStepper.h: In function 'uint8_t get_status_response(TMC2130Stepper&)':
    C:\Users\Ludy\Documents\Arduino\libraries\TMCStepper\src/TMCStepper.h:334:11: error: 'uint8_t TMC2130Stepper::status_response' is protected
    uint8_t status_response;
           ^
    C:\Users\Ludy\AppData\Local\Temp\arduino_build_799868\sketch\src\feature\tmc_util.cpp:57:72: error: within this context
     static uint8_t get_status_response(TMC2130Stepper &st) { return st.status_response & 0xF; }
                                                                        ^
    In file included from 
    C:\Users\Ludy\AppData\Local\Temp\arduino_build_799868\sketch\src\feature\tmc_util.h:28:0,
                 from 
    C:\Users\Ludy\AppData\Local\Temp\arduino_build_799868\sketch\src\feature\tmc_util.cpp:27:
    C:\Users\Ludy\Documents\Arduino\libraries\TMCStepper\src/TMCStepper.h: In function 'TMC_driver_data get_driver_data(TMC2130Stepper&)':
    C:\Users\Ludy\Documents\Arduino\libraries\TMCStepper\src/TMCStepper.h:334:11: error: 'uint8_t TMC2130Stepper::status_response' is protected
    uint8_t status_response;
           ^
    C:\Users\Ludy\AppData\Local\Temp\arduino_build_799868\sketch\src\feature\tmc_util.cpp:69:27: error: within this context
       data.is_error = (st.status_response & DRIVER_ERROR_bm) >> DRIVER_ERROR_bp;

@teemuatlut I think it's a bug of the library

I compile with PIO and it compiled.

Marlin start and after to run M502/M500:

Printer is now online.
echo:Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Oct  3 2018
echo: Free Memory: 2563  PlannerBufferBytes: 1648
Error:Field planner_extruder_advance_K mismatch.
[ERROR] Error:Field planner_extruder_advance_K mismatch.

Error:EEPROM datasize error.

LIN_ADVANCE is disable.

teemuatlut, 2018-10-03:
Install the latest commit of the library.
This has been fixed already but I haven't made a release tag for Arduino IDE yet.

thinkyhead, 2018-10-03:
Try the latest `bugfix-2.0.x` to see if your EEPROM error is fixed.

Ludy87, 2018-10-03:
I have tried and it is fixed! Thanks!!

closed 2018-10-03 by Ludy87