MarlinFirmware/Marlin #12000: my local github clone show smore than 1000 changes (in both directions) since  last week

autonumous, 2018-10-04:
Im not git expert, but this driving me mad, something similar happened last week, and now again.

I‘ve cloned this Marlin repo locally, and i keep  branch 1.1.x & bugfix-1.1.x updated.
 I have my own branch (e.g. MyPrinter-1.1.9 & MyPrinter-BF-1.1.x)  which i use  for my own printer customisations.  I periodically merge the changes from  bugfix-1.1.x into MyPrinter-BF-1.1.x, and 1.1.x into MyPrinter-1.1.9.

today my local 1.1.x shows over 1000 different commits in each direction.  
eg 

```
 git status
On branch 1.1.x
Your branch and 'upstream/1.1.x' have diverged,
and have 1010 and 1181 different commits each, respectively.
```

i’ve looked at a couple of the commits

from my git log, one of the most recent commits is 
https://github.com/MarlinFirmware/Marlin/tree/caa601a

i’ve traced this same commit in the current github repo to: 
https://github.com/MarlinFirmware/Marlin/tree/6afe5ab

for caa601a the modification date for the majority of the files shows “12 days ago”.  

There shouldn’t be two commits for the same change should there?   Its like something “forked” and then been reset....

last week or so, something similar occurred,  Resulting in me discarding all the identified  changes and then updating again.  This then had the consequence of screwing up my own customised branch as my edits appeared to be older requiring me to manually update every change/update that I've made.

Now the same has happened.  Whats going on?   Not seen any other reports, which makes me wonder it’s something i’ve done, i’ve following the same update/merge process for a number of years with no issue.

Roxy-3D, 2018-10-04:
Wow!!!   Post #12000 !  

Visual Diff tools can accomplish much of what Git does.    Usually it isn't too hard to resync the stuff you have added to a fresh copy of the main branch.

thinkyhead, 2018-10-04:
We occasionally rebase some branches as part of general cleanup. As a result your fork's (`origin`'s) copy of the `upstream` branches will sometimes go out of sync. There are some good guidelines to follow to help mitigate the problems that this may cause for your downstream development:

1. Never modify your fork's copies of the `upstream` branches (`1.1.x`, `bugfix-1.1.x`, and `bugfix-2.0.x`). Always create a new branch when doing new work.
2. To sync your fork's copies of the `upstream` branches with ours, use `git reset`, as in:
    ```
     git fetch upstream
     git checkout 1.1.x
     git reset --hard upstream/1.1.x
     git push -f origin
    ```
3. Get familiar with `git rebase`, `git diff`, `git apply`, and `git cherry-pick` to be able to adapt ongoing or old work to evolving `upstream` branches.
4. Always start new work using (at this time) a copy of the `bugfix-2.0.x` branch, as this is the only branch currently accepting new features. Reiterating the first point, never add on to any of your downstream branches. Make a copy as a new branch and work with the copy.

We're increasingly moving to a "rebase" (rather than "merge") workflow, so it will definitely be helpful going forward to get familiar with `git rebase` and other "git voodoo" to adapt your ongoing work to our evolving and occasionally-rebased branches.

autonumous, 2018-10-04:
Thanks guys,   Good advice, i’ll Adjust my workflow.  
 I’ve always followed points 1 and 4, and let ‘github desktop’  deal with merging/updating etc. So not sure what it does under the hood.  But  based on what you say, I can see can see a git pull could cause issues. 
I’ll start doing it manually now....

Thanks

closed 2018-10-04 by autonumous