MarlinFirmware/Marlin #12001: PlatformIO Upload to Re-Arm fails

Squid116, 2018-10-04:
### Description

PlatformIO Upload to Re-Arm fails

### Steps to Reproduce

1. PlatformIO build -> works
2. Copy .bin to sd card, reboot and all works
3. Trying to upload via platformio fails, see error log below.
4. I have uninstalled vscode, platformio, updated python no luck
5. Same result between autobuild and platformio menu, or verbose upload.
6. Tried atom and platformio, same result.
7. blew away and re-forked my repo: https://github.com/Squid116/Marlin/tree/bugfix-2.0.x no luck

**Expected behavior:** [What you expect to happen]
Uploads latest code to re-arm

**Actual behavior:** [What actually happens]
Fails to upload.

#### Additional Information
```
> Executing task: C:\Users\Chris\.platformio\penv\Scripts\platformio.exe run --verbose --target upload <

Processing LPC1768 (board_upload.maximum_ram_size: 32768; board_build.f_cpu: 100000000L; build_flags: !python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py -fmax-errors=5, -g, -ggdb; lib_deps: CMSIS-LPC1768, https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip, https://github.com/teemuatlut/TMCStepper.git; debug_tool: custom; build_unflags: -Wall; extra_scripts: Marlin/src/HAL/HAL_LPC1768/debug_extra_script.py, Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py, Marlin/src/HAL/HAL_LPC1768/upload_extra_script.py; platform: nxplpc@<3.4.0; debug_server: C:\Program Files (x86)\SEGGER\JLink_V618d\JLinkGDBServerCL.exe, -select USB, -port 2331, -device LPC1768, -if JTAG, -speed auto, -noir; board: lpc1768; lib_extra_dirs: frameworks; monitor_speed: 250000; lib_ldf_mode: off; src_filter: +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared> +<src/HAL/HAL_LPC1768>)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Storage: c:\git\Marlin\.piolibdeps
Looking for CMSIS-LPC1768 library in registry
Warning! Library `{'requirements': None, 'name': 'CMSIS-LPC1768'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'CMSIS-LPC1768'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
LibraryManager: Installing dev
U8glib-HAL @ 0.4 is already installed
LibraryManager: Installing TMCStepper
TMCStepper @ 5d3abfe is already installed
PLATFORM: NXP LPC > NXP mbed LPC1768
SYSTEM: LPC1768 100MHz 32KB RAM (512KB Flash)
DEBUG: CURRENT(custom) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(soft)
Collected 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CMSIS-LPC1768> 0.0.0 (c:\git\Marlin\frameworks\CMSIS)
|-- <U8glib-HAL> 0.4 [https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip] (c:\git\Marlin\.piolibdeps\U8glib-HAL_ID1932)
|-- <TMCStepper> 0.2.0 #5d3abfe [git+https://github.com/teemuatlut/TMCStepper.git] (c:\git\Marlin\.piolibdeps\TMCStepper)
upload disk:  F:
MethodWrapper(["checkprogsize"], [".pioenvs\LPC1768\firmware.elf"])
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  30.5% (used 10004 bytes from 32768 bytes)
PROGRAM: [===       ]  30.7% (used 160812 bytes from 524288 bytes)
.pioenvs\LPC1768\firmware.elf  :
section              size        addr
.text              160092       16384
.ARM.exidx              8      176476
.data                 712   268435656
.bss                 9292   268436368
.heap                2048   268445664
.stack_dummy         3072   268445664
.ARM.attributes        41           0
.comment              126           0
.debug_info       1298943           0
.debug_abbrev      199829           0
.debug_loc         120288           0
.debug_aranges      10800           0
.debug_ranges       22184           0
.debug_line        204025           0
.debug_str          83967           0
.debug_frame        33952           0
.stab                  60           0
.stabstr              118           0
Total             2149557
echo 'h\nloadfile c:\git\Marlin\.pioenvs\LPC1768/firmware.hex\nr\nq\n' > c:\git\Marlin\.pioenvs\LPC1768/aux.jlink
The system cannot find the file specified.
*** [upload] Error 1
================================================================= [ERROR] Took 11.61 seconds =================================================================

========================================================================= [SUMMARY] =========================================================================
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment at90usb1286_cdc             [SKIP]
Environment at90usb1286_dfu             [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1768                     [ERROR]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment STM32F4                     [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
================================================================= [ERROR] Took 11.67 seconds =================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

p3p, 2018-10-04:
> upload disk:  F:

Is this detected upload disk correct? I just performed a clean clone build/upload and it detected the disk and copied the file correctly,

Squid116, 2018-10-04:
Just changed drive letters to be sure, platformio picked up the new drive letter, but still failed.

Squid116, 2018-10-04:
Nuked (massive) .platformio directory under my appdata folder.  Re downloaded everything and still no luck :(

Squid116, 2018-10-04:
Commenting out the following:
```
debug_tool        = custom
debug_server      =
  C:\Program Files (x86)\SEGGER\JLink_V618d\JLinkGDBServerCL.exe
  -select USB
  -port 2331
  -device LPC1768
  -if JTAG
  -speed auto
  -noir
```

And removing the debug_extra_script git it working... I suppose I should look into getting those tools working, but not tonight.

p3p, 2018-10-04:
That shouldn't interfere with a normal build / upload (and doesn't for me), I don't use them either. If they can interfere they will be removed as they serve no purpose for normal users.

CAP1Sup, 2018-10-04:
How long is the filename? I had an issue where the file path was too long and Marlin wouldn't compile (with Platform.io only).

Squid116, 2018-10-05:
@CAP1Sup not that long:

`c:\git\Marlin\.pioenvs\LPC1768/firmware.hex`

But it is right in the assertion that it doesn't exist.

thinkyhead, 2018-10-07:
- First partition is 32GB or less?
- SDCard is named 'REARM'?

It might just be an issue with PlatformIO. Have you posted in the PlatformIO Community Forum?

closed 2019-01-06 by Squid116