MarlinFirmware/Marlin #12003: Reprap Discount Full Graphics Smart Controller Display is Garbled

dcwalmsley, 2018-10-04:
So here I am again.

Downloaded the latest 1.1.9 bugfix since the my last 1.1.8 bugfix release and developed an issue that has been an annoyance for me and I'm sure for some with Reprap Discount Full Graphics Smart Controller Display coming up garbled after flashing firmware.  My LCD is the red version (12864) of the Reprap boards.

I've attached two images of the LCD display and the zipped folder of my modifications to your latest BugFix.

My printer is a HyperCube using a Einsy Rambo board.
I'm also using U8glib 1.18.0 currently and have tried 1.19.1 with no success on either library.
The compiler I'm using is an Arduino 1.8.7 program.

[Marlin-1.1.9 Release 10-3-2018.zip](https://github.com/MarlinFirmware/Marlin/files/2446462/Marlin-1.1.9.Release.10-3-2018.zip)
[image]
[image]

Did I miss something in updating the firmware?

Appreciate all that you great folks do and look forward to some help fixing this problem.

Best Regards,
Doug Walmsley

larsnordstrom, 2018-10-04:
You have the connector on the display in the wrong orientation, you need to pry the  two plastic connector off on the display and rotate them 180 degrees. That'll fix your problem.

dcwalmsley, 2018-10-04:
Thanks for the insight but that was done back in build 1.1.8 and it worked as you described.  So there should be no difference now that I I have upgraded to the latest version.  This is a firmware issue.

larsnordstrom, 2018-10-04:
Ok, then you need to add a condition to ultralcd_st7920_u8gkib_rrd.h
locate line 56 and make sure it reads:

#elif MB(MINIRAMBO) || MB(EINSY_RAMBO)

AnHardt, 2018-10-04:
Looks like a timing problem. Not like a connection problem. Increase delays.

dcwalmsley, 2018-10-04:
@larsnordstrom 

Thanks!  That addition fixed my issue.

[image]

closed 2018-10-07 by thinkyhead

Kenzox13, 2018-11-09:
> @larsnordstrom
> 
> Thanks! That addition fixed my issue.
> 
> [image]

How did you solve it?
I have the same problem with a tornado and marlin tevo 1.1.9.

I define in config.h, as I understood,but it still does not work, it looks fuzzy. What do I do wrong?

#define ST7920_DELAY_1 DELAY_NS(0)
#define ST7920_DELAY_2 DELAY_NS(250)
#define ST7920_DELAY_3 DELAY_NS(0)

Forgive my english... heheheh

thinkyhead, 2018-11-09:
Try some longer delays, for example:
```cpp
#define ST7920_DELAY_1 DELAY_NS(250)
#define ST7920_DELAY_2 DELAY_NS(250)
#define ST7920_DELAY_3 DELAY_NS(250)
```

Kenzox13, 2018-11-09:
Thanks, I have already achieved it.

#define ST7920_DELAY_1 DELAY_NS (0)
#define ST7920_DELAY_2 DELAY_NS (400)
#define ST7920_DELAY_3 DELAY_NS (0)

I went up from 50 in 50 until the problem was corrected.

Likewise, thank you very much.

lb4207, 2019-10-07:
> @larsnordstrom
> 
> Ok, then you need to add a condition to ultralcd_st7920_u8glib_rrd.h
> locate line 56 and make sure it reads:
> #elif MB(MINIRAMBO) || MB(EINSY_RAMBO)

How do I edit ultralcd_st7920_u8glib_rrd.h in Arduino?
I can click on the dropdown list on the right of the toolbar and select the file from the list but nothing appears. Where do I go from there?

offtherails2010, 2019-10-09:
@thinkyhead &
@AnHardt 

i just wanted to say a BIG thanks to you both, the delay setting also worked for me using MEGA2560/RAMPS/RRD Full Graphic Smart Controller (with buzzer, STOP Button, POT & Encoder, RED PCB By www.robotlinking.com)

Racking my brain over this for days lol

Thanks LOADS guys, really grateful !

gazehound, 2019-10-19:
the lines
#define ST7920_DELAY_1 DELAY_NS (0)
#define ST7920_DELAY_2 DELAY_NS (400)
#define ST7920_DELAY_3 DELAY_NS (0)

were do you put them at?

lb4207, 2019-10-21:
> 
> 
> the lines
> #define ST7920_DELAY_1 DELAY_NS (0)
> #define ST7920_DELAY_2 DELAY_NS (400)
> #define ST7920_DELAY_3 DELAY_NS (0)
> 
> were do you put them at?

From post above by larsnordstrom
Ok, then you need to add a condition to ultralcd_st7920_u8glib_rrd.h
locate line 56 and make sure it reads:

#elif MB(MINIRAMBO) || MB(EINSY_RAMBO)

You'll find that line already there and the 3 lines you refer to are 57, 58 and 59

darap89, 2019-10-21:
> From post above by larsnordstrom
> Ok, then you need to add a condition to ultralcd_st7920_u8glib_rrd.h
> locate line 56 and make sure it reads:
>
> #elif MB(MINIRAMBO) || MB(EINSY_RAMBO)
>
> You'll find that [...]

i have the same problem but i have an ramps 1.4 borad,where do i change the delay)?

thinkyhead, 2019-10-21:
> i have the same problem but i have an ramps 1.4 borad,where do i change the delay)?

If you want you can define your own set of delays in `Configuration.h`.

darap89, 2019-10-22:
Hi again.
I searched in config.h and I didn't find such an comment I could change but I went back into (ultralcd_st7920_u8glib_rrd.h)tab
And there is 3 lines where it says you can define your own.
I changed mine to:
#define ST7920_DELAY_1 DELAY_NS (0)
#define ST7920_DELAY_2 DELAY_NS (400)
#define ST7920_DELAY_3 DELAY_NS (0)

And everything works fine now.big thanks from me anyway

gazehound, 2019-10-22:
i found them lines and changed them to what is here but i seen no change to the screen

darap89, 2019-10-22:
@gazehound 
I can't explain it right as my English isn't that good but did you remember to uncommint the 3 lines after you have changed the numbers

gazehound, 2019-10-22:
yes i did remembered to do this time :). just tring to learn what each controls so know how to play with them

lb4207, 2019-10-25:
I've tried 0, 400, 0 and 250, 250, 250 but nothing changes.
These are my lines starting at #56
#elif MB(MINIRAMBO) || MB(EINSY_RAMBO) || MB(EINSY_RETRO)
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(400)
  #define CPU_ST7920_DELAY_3 DELAY_NS(0)

Can anyone suggest what I have wrong?

darap89, 2019-10-25:
@lb4207 what board do you have mate ?

lb4207, 2019-10-25:
MKS GEN L with RepRapDiscount Full Graphic Smart Controller.
I didn't have this problem before the Arduino 1.8.10 upgrade.
lb4207

darap89, 2019-10-25:
@lb4207  same her is just have an ramps 1,4.

open your autdino ang go to the tab called (ultralcd_st7920_u8glib_rrd.h)tab
And there is 3 lines where it says you can define your own.
I changed mine to:
#define ST7920_DELAY_1 DELAY_NS (0)
#define ST7920_DELAY_2 DELAY_NS (400)
#define ST7920_DELAY_3 DELAY_NS (0)

the tree line is at the start of the page,that's them you have to change since you have another board.when you change them to this they will overide the other delay settings.
I changed mine to what you see above and everything is working great With the screen.

ktraglin, 2019-11-14:
**larsnordstrom**'s instructions worked for me with some modification:
`Ok, then you need to add a condition to ultralcd_st7920_u8glib_rrd.h
locate line 56 and make sure it reads:

#elif MB(MINIRAMBO) || MB(EINSY_RAMBO)`

Since my board is an MKS Gen (v1.4), I added it to the end of this line, so it reads:

#elif MB(MINIRAMBO) || MB(EINSY_RAMBO) || MB(MKS_GEN_13)

This solved this issue for me.  Thanks **larsnordstrom**

lb4207, 2019-11-14:
> I've tried 0, 400, 0 and 250, 250, 250 but nothing changes.
> These are my lines starting at #56
> #elif MB(MINIRAMBO) || MB(EINSY_RAMBO) || MB(EINSY_RETRO)
> #define CPU_ST7920_DELAY_1 DELAY_NS(0) [...]

I found that if I added those lines to configuration.h it worked but they were not recognised in ultralcd_st7920_u8glib_rrd.h

ValentynRu1, 2019-11-22:
> I found that if I added those lines to configuration.h it worked but they were not recognised in ultralcd_st7920_u8glib_rrd.h

Configuration.h

#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 
#define ST7920_DELAY_1 DELAY_NS(0) 
#define ST7920_DELAY_2 DELAY_NS(400) 
#define ST7920_DELAY_3 DELAY_NS(0)

It helped me. Thanks for telling me where to look for the problem

Cristaux, 2019-11-25:
I extended the queues of my screen 1 meter and same problem of display, my solution:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

#define ST7920_DELAY_1 DELAY_NS (0)
#define ST7920_DELAY_2 DELAY_NS (250)
#define ST7920_DELAY_3 DELAY_NS (250)
good luck...

RawSmokeTerribilus, 2019-12-15:
Hi and thanks to every body!!! that fix my issue, used a mixed values (250,400,250) and it works perfectly.
I just wonder why after almost a year working fine it start glitching like that. the only thing that i did was to install my marlin 1.9 from always (whit my stuffs and configs) with a freshly updated Arduino IDE... but my libraries whas the same and everything was updated too... SOo i start wondering if i have something being broken on my controller...

StockyJ, 2019-12-17:
I have this issue with Marlin V2 can anyone advise how to modify V2 with these settings.

darap89, 2019-12-17:
I have just updated to V2 and No issue,but o guess it's the same folder you gave to go thru.

Få Outlook til Android<https://aka.ms/ghei36>

snail007, 2019-12-20:
> I extended the queues of my screen 1 meter and same problem of display, my solution:
> #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> 
> #define ST7920_DELAY_1 DELAY_NS (0)
> #define ST7920_DELAY_2 DELAY_NS (250)
> #define ST7920_DELAY_3 DELAY_NS (250)
> good luck...

thx,it solved my problem

Hakancosgun, 2020-02-13:
thanks for all 😃

shabgarde, 2020-02-15:
> Configuration.h
>
> #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> #define ST7920_DELAY_1 DELAY_NS(0)
> #define ST7920_DELAY_2 DELAY_NS(400)
> #define ST7920_DELAY_3 DELAY_NS(0)
>
> It helped me. [...]

tanx, it solved my problem too

dgpower, 2020-05-01:
结合上面各位的讨论，我找到一个能用的方法：
1\先看看 Configuration.h里面的主板型号，我这里是BOARD_RAMPS_14_EFB
     #define MOTHERBOARD BOARD_RAMPS_14_EFB 
2、打开ultralcd_st7920_u8glib_rrd.h，找到下面这一行代码（一般是在第56行）
    #elif MB(MINIRAMBO) || MB(EINSY_RAMBO) || MB(EINSY_RETRO)
     修改成如下格式
    #elif MB(MINIRAMBO) || MB(EINSY_RAMBO) || MB(EINSY_RETRO)|| MB(RAMPS_14_EFB)
    它的作用就是让下面的三行延时代码其作用，其他延时都不用改。

3、完整的代码发出来，供大家参考
   #elif MB(MINIRAMBO) || MB(EINSY_RAMBO) || MB(EINSY_RETRO)|| MB(RAMPS_14_EFB) 
      #define CPU_ST7920_DELAY_1 DELAY_NS(0)
      #define CPU_ST7920_DELAY_2 DELAY_NS(250)  
      #define CPU_ST7920_DELAY_3 DELAY_NS(0)

avilab84, 2020-05-10:
DAMN, why is this not fixed ?

shabgarde, 2020-05-10:
Hello
This solution worked perfectly for me...
I pased this 3 line code in configuration.v

thinkyhead, 2020-05-12:
> DAMN, why is this not fixed ?

It is patched in mainline Marlin. 🦦

SpdPnd98, 2020-05-17:
> I extended the queues of my screen 1 meter and same problem of display, my solution:
> #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> 
> #define ST7920_DELAY_1 DELAY_NS (0)
> #define ST7920_DELAY_2 DELAY_NS (250)
> #define ST7920_DELAY_3 DELAY_NS (250)
> good luck...

Thanks! This solved my problem!!!